LinuxCNC/linuxcnc #2716: multiple parallel linuxcnc sessions

rmu75, 2023-10-30:
As @rene-dev mentioned to me it is currently not possible to run tests in parallel, mainly because there can only be one linuxcnc instance running at a time. After investigating for a bit and looking at old branches in linuxcnc and machinekit I think such a feature could be realized with little effort.

CMS/NML uses SYSV shared memory buffer and semphores ("ipcs"), those resources are identified by a key (32bit integer) and those keys are derived from specification in the .nml file, usually those numbers are 1001-1005 or so.

HAL uses its own shared memory segments identified by key 0x48484c34 ("HHL\"") and 0x48414c32 ("HAL "), halscope involves another one with hardcoded key, for testing purposes that should not be relevant.

So in order to run tests in parallel, either the sessions have to be isolated in some way such that each session gets it's own isolated SYSV ipc namespace (that would involve some linux namespace trickery and probably need root) or there has to be a mechanism to change the keys to non-overlapping unique values. AFAIK that should also be sufficient for parallel test runs in user space.

I propose using an environment variable like "LINUXCNC_INSTANCE" containing a number that is used as a kind of offset, e.g. first instance uses 0x48414c32 and 1001/1002/1003, second instance uses 0x49414c32 and 1011/1012/1013 and so on.

Please comment if I missed something. If this feature would be welcome and no major obstacles turn up I plan on implementing it.

andypugh, 2023-10-31:
Would this only be for testing, or do you see it being useful for running (for example) a second toolpath on a multi-axis lathe? 
Would the separate instances have any way to communicate with each other? 
(A simple soution to this might be a HAL component that could create pins in two share memory spaces, but there are probably cleverer ways to do it)

rmu75, 2023-10-31:
Yeah could be, but synchronizing two instances will be painful.

I already have something working but get some very strange effects, seems I still miss a few locations.

rmu75, 2023-10-31:
> Would this only be for testing, or do you see it being useful for running (for example) a second toolpath on a multi-axis lathe? Would the separate instances have any way to communicate with each [...]

I think for that use-case it would be cleaner to use one HAL instance and make task/etc... multi instance capable.

andypugh, 2023-10-31:
> I think for that use-case it would be cleaner to use one HAL instance and make task/etc... multi instance capable.

This would need a way to fix HAL pin name clashes. But as all pins are created by the same function, it could be done there. 
0:motion.motion-enable and 1:motion.motion-enable maybe (with 0: being assumed if omitted)

rmu75, 2023-10-31:
... and some way to discern multiple command / error / status NML channels and motmod instances.

rmu75, 2023-10-31:
how to parallelize tests? GNU parallel? https://www.gnu.org/software/parallel/