LinuxCNC/linuxcnc #2719: Tool Database Interface not working

byunchov, 2023-10-30:
The tool database interface appears to be broken.
I was using `[EMCIO]TOOL_TABLE` as my primary tool data source but I'm currently in the process of migrating to a SQLIte3 database and decided to test my tool database interfaces for  `[EMCIO]DB_PROGRAM` with no luck. It took me a good hour or so to figure that there was something wrong in LCNC rather than my interface. 
As starting point I used `configs/sim/axis/db_demo/db_nonran.ini` sample config to get started.

### Here are the steps I follow:

 1. Copy the sample config to my local `~/linuxcnc/configs/` directory.
 2. Provided the path to database interface script in `[EMCIO]DB_PROGRAM` in the `ini` file.
 3. Saved the file and copied the `ini` file path.
 4. Open a terminal and launched the config with `linuxcnc <ini_path>` to observe the `stdout` for error messages form my script.

#### This is what I expected to happen:
```
Starting LinuxCNC...
====Waiting for v2.1 version reply from <script_path>
...
====Connected to  <script_path>
GET:   T2201 P302 D45.00 X0.0000 Y0.0000 Z2.6868 C0.0000
GET:   T1 P405 D30.00 X0.0000 Y0.0000 Z0.0000 C0.0000
GET:   T2 P406 D14.00 X0.0000 Y0.0000 Z0.0000 C0.0000
GET:   T4001 P407 D34.00 X0.0000 Y0.0000 Z0.0000 C0.0000
GET:   T5002 P408 D12.00 X0.0000 Y0.0000 Z0.0000 C0.0000
GET:   FINI
```
#### This is what happened instead:
```
Starting LinuxCNC...
...
emc/task/taskclass.cc 168: can't load tool table.
```

### It worked properly before this:
I was running source build binaries from commit 2df5e87 and decided to upgrade to the latest 4edef86 because I was facing random shutdowns of the `Axis` GUI due to buffer overflow of OpenGL. Sadly I didn't  save the log files (I had to stop the spindle from running as the crash cut the connection to the VFD and left it free-running).
So far I haven't met this nasty error but not `[EMCIO]DB_PROGRAM`. 
I also did some tweaking to the interpreter internals to add a binary function that allows retrieving tool diameter in gcode by issuing `TD[toolno]`. At first I thought I mesed something up but I pulled again from master branch and rebuild the binaries. The result i still the same. I'm not that familiar with the internals of EMC and I really could use some help with this issue.  

## Information about my hardware and software:

 * Linux distribution:
 ```
Distributor ID:	Linuxmint
Description:	LMDE 5 (elsie)
Release:	5
Codename:	elsie
```
 * Kernel version:
```
Linux cnc 5.10.0-26-rt-amd64 #1 SMP PREEMPT_RT Debian 5.10.197-1 (2023-09-29) x86_64 GNU/Linux
```
 * I am running ...
   * [x] A binary I built myself
 * I am using this LinuxCNC version (`scripts/get-version-from-git`):
 ```
v2.10.0-pre0-2305-g4edef86857
```
 * I am using `AXIS` as GUI to test this config and run my machine.
 * I am using no specific hardware for this simulation.

byunchov, 2023-11-01:
Thank you, @andypugh !
I did git comare on 2df5e87..4edef86and and went trough all of the changes introduced in the recent revision of master branch. It was a bit tedious but your suspicions mught turn out right. I don't know very well how everything is tied together I just did a quick code review. 
I will revert back to my build of 2df5e87 and probably will introduce the changes to the interpreter I've made to use the binary operator for retrieving tool diameter from the tool table.
If I can help with something I'd be more than happy to do so as someone else might be in the same place as me several days ago (debugging a piece of code that was properly working for no reason).

andypugh, 2023-11-01:
Git bisect would have reduced the tediousness somewhat. It's a very useful tool. 
Basically it does an automatic binary search, and you tell it at each point whether that particular build is "good" or "bad". Eventually (and in surprisingly few steps) you home in on the commit that caused the problem.

byunchov, 2023-11-01:
I will do git bisect when I get some free time. Maybe this will turn out to be useful.
I'm not that familiar with git tools, despite the fact that I'm software engineer 😔 I forked Lcnc to be sure that I don't mess something up. I wanted to push the change to the interpreter I made but didn't know what to do.

PR #2733 (2023-11-06): Fixes db_demo simulation and adds back DB_PROGRAM functionality
The should add back the missing functionality to use DB_PROGRAM, it seems this was forgotten in 7ebe7d91fced459729c84a4265556692010379c9. This also adds two error messages and  initializes variables in the header file.

The simulation `db_demo`, both random and nonrandom should be fixed with this PR.

Fixes #2719
closed 2023-11-06 by rene-dev