LinuxCNC/linuxcnc #2726: Carousel stopped working concurrently with a Buildbot 2.10 update

BSHoekstra, 2023-11-03:
The problem that I described in issue #2025 has resurfaced again with a recent 2.10 buildbot update.  Any help you can provide would be appreciated. Thank you, Barry :)

Here are the steps that I follow to reproduce the issue:

Carousel WORKS properly in 2.9.1-25-ga3295d4c8.  Carousel works properly by indexing the ATT to the tool selected.  The tool change command can be issued multiple times and each time the ATT will index properly to the new tool selected.

Carousel DOES NOT work properly in  Buildbot 2.10.0-pre0-2330-g292511db43.  Carousel works properly the 1st time the Tool Change command is issued, but then Carousel will NOT index the ATT with additional tool change commands. There are no error messages. However, graphically, Axis shows that a tool change was executed.  Note when I exit from Linuxcnc and restart it, the ATT will again index 1 time correctly and then not accept additional tool change commands.

Please reference issue #2025 for hardware configuration and for the hal script files.  None of this has been changed on my cnc lathe. 

Using the Hal Scope configuration file provided by Andy in issue #2025, I have attached hal scope log files for your reference.
Both log files start with Tool #4 and then tool change commands are issued to index to Tool #3, then Tool #4, then Tool #5. 

The attahced "working" hal scope log file was captured using 2.9.1-25-ga3295d4c8.
[halscope_working.csv](https://github.com/LinuxCNC/linuxcnc/files/13246535/halscope_working.csv)

The attached "not_working" hal scope log file was captured using Buildbot 2.10.0-pre0-2330-g292511db43.
[halscope_not_working.csv](https://github.com/LinuxCNC/linuxcnc/files/13246537/halscope_not_working.csv)

Please let me know if any addtional information is needed or if I can be of any help with additional testing.

 Thank you in advance for your help with this,
 
Barry :)

andypugh, 2024-01-07:
Sorry, I have been sitting on this. Is there still a problem? ie, nobody has fixed it by accident?

BSHoekstra, 2024-01-08:
No worries on the delay.   I just appreciate you finding time to look into the issue to resolve it.  In the meantime, I have been using Version 2.9.  

I temporarily reloaded Buildbot 2.10 two weeks ago to test if the issue had been resolved, but unfortunately it was not, so I went back to 2.9.  Please note that I have not yet tested to see if issue is in the Carosel.so file but I can do that tonight by dropping the 2.10 Carosel.so file into version 2.9 if that information would be helpful to you.  Also, I am still using Buster and have not yet upgraded to Bookworm, but I can do that if you need me too.

Thank you,

Barry :)

andypugh, 2024-01-08:
Do you know what date your 2.9 version is? I haven't had much luck so far using the hash (ga3295d4c8) to find that out. 
There isn't much in master:
https://github.com/LinuxCNC/linuxcnc/commits/master/src/hal/components/carousel.comp
That isn't also on 2.9
https://github.com/LinuxCNC/linuxcnc/commits/2.9/src/hal/components/carousel.comp
The only difference seems to be the correction of a typo in the documentation section.

BSHoekstra, 2024-01-09:
Per your request, these are the tests I performed tonight,

1)  Carousel worked in Linuxcnc 2.9.1-60-gf0eb99fd4.  So, I copied and saved the V2.9 carousel.so file.

2)  I then deleted Linuxcnc V2.9  and loaded Linuxcnc 2.10.0-pre0-2617g0487cb3269.  Carousel rotated the tool 1 time and then the ATC would not respond to additional tool change commands.

3)  With Linuxcnc V2.10.0 loaded, I replaced the V2.10 carousel.so file with the V2.9 carousel.so file.  Carousel rotated the tool 1 time and then the ATC would not respond to additional tool change commands.

Wow, so whatever is causing the issue, I'm guessing is outside the carousel.so file.  That might be hard to locate.  Good luck!  

Please let me know if I can be of any further help.

Thank you,

Barry :)

BSHoekstra, 2024-06-01:
FYI.....I loaded buildbot 2.10 again today just to see if by chance the Carousel issue had been resolved but found that the issue is still there as described in this post.   

Thank you,  Barry :)

andypugh, 2024-06-28:
There is a very small chance that this has been fixed by https://github.com/LinuxCNC/linuxcnc/pull/2997
Is it possible to demonstrate the issue with the sim-axis-vismach-vmc_toolchange config? That would make a Git Bisect a relatively easy way to spot the troublesome commit.

andypugh, 2024-06-28:
So, back to the question about whether the problem can be observed without access to your actual machine.

BSHoekstra, 2024-06-28:
Hello Andy.  FYI….It will be 2 weeks before I’ll be back in my cnc machine shop to be able to provide you with feedback.  Sorry for this delay.  

Barry

BSHoekstra, 2024-07-08:
In response to the request in issue #2025, Can you check that you get stuck with enable false, and ready not set, and with the position not matching the pocket number?

I just loaded: 2.10.0-pre0-3309-gfd6a133f8b

With enable false, and ready not set, and with the position NOT matching the pocket number, the auto-tool changer indexes 1 time to the correct tool and then it will not respond to the next tool change command; however, the Axis graphical display changes to show the correct tool shape/orientation/etc. every time a new tool change command is issued.

With enable false, and ready not set, and with the position matching the pocket number the auto-tool changer indexes 1 time to the correct tool and then it will not respond to the next tool change command; however, the Axis graphical display changes to show the correct tool shape/orientation/etc. every time a new tool change command is issued.

Thank you,

Barry :)

andypugh, 2026-01-15:
I haven't actually forgotten this, but neither have I found the time to look at it. 

I assume you still see this issue?