LinuxCNC/linuxcnc #2744: Active Gcodes list does not properly reflect applied G52 offsets

Sigma1912, 2023-11-11:
1. Command G52 X100 -> Offset is applied in DRO and active GCodes will show 'G52'
2. Command M2  (or M30)  -> Offset remains applied in DRO but active GCodes no longer shows 'G52'

This is from the documentation:

[image]

If the G52/G92 offset remains applied after  M2/M30 then 'G52'/'G92' should not be removed from the active gcodes list.

This is using a recent build of version 2.9

Sigma1912, 2024-03-03:
Looking at this again it seems that the actual issue here is that G52 persists after M2/M30 when it should really be cleared.

Sigma1912, 2024-03-12:
Seems that G52/G92 handling in LinuxCNC does not follow NISTs RS274NGC standard.
While in the standard the offset  is deactivated on 'm2'/'m30' and the offset values are made 'persistent' by storing them, LinuxCNC will not deactivate G92/G52 offsets on 'm2'/'m30' and both values AND activation state are 'persistent'.
https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374

[image]  
Looking at the number of tests failing when changing this makes me think that this must have been done intentionally.

andypugh, 2024-03-16:
Feel free to re-open this if #2935 wasn't actually the fix

closed 2024-03-16 by andypugh