LinuxCNC/linuxcnc #2745: Could Active Gcodes list reflect the active work offset (g5x) based on parameter[5220]?]

Sigma1912, 2023-11-11:
I'm using python remaps and because of the caveat of not being able to issue a 'self.execute()' command after a 'yield INTERP_EXECUTE_FINISH' I use emccanon commands to change active work offsets eg:

```
self.params[5220] = 9
emccanon.SET_G5X_OFFSET(9,P[0],P[1],P[2],0, 0, 0, 0, 0, 0, )
```

While this will work and is totally legit it produces a problem in the guis:

Axis GUI: Will display the correct offset values and index in the DRO but the active GCode list will not update and show the wrong offset index.

Gmoccapy: Will display the correct offset values but the index in the DRO and the active Gcode list will show  the wrong offset index.

I'm not sure about this as I have not found the code responsible for it yet but it seems  that the active Work Offset in status.gcodes list gets only updated by GCodes (G54 ... G59.3) issued through the interpreter.
I wonder if the active Work Offset in the status.gcodes list could be made to reflect the actual value of parameter 5220 instead. 

Ie something like this (although it is probably done in c/cpp):

```
            code = self.status.g5x_index
            if code >= 1 and code <= 6:
                return "G%s" % int((code + 53))
            elif code > 6 and code <= 9:
                return "G59.%s" % (code - 6)
```

Sigma1912, 2023-11-12:
Actually I just noticed that switching offset index with emccanon commands can lead to inconsistencies not only with the index value but also with the offset values as apparently it is not always the values stored in the numbered parameters that are used but others that are stored elsewhere. I guess in the gui until that gets closed.

Sigma1912, 2023-11-12:
I clearly didn't understand the way things work so I close this.

Sigma1912, 2023-11-15:
Reopened with a clear bug description

Sigma1912, 2023-11-15:
Workaround:

With this in the [RS274NGC]-section of the ini:

ON_ABORT_COMMAND = o <on_abort> call

and this in the on_abort.ngc:

```

o<on_abort> sub
g59
g54
o<on_abort> endsub
m2
```

Notes: 
- We need to switch to an offset-index other than g54 first for this to work (eg 'g59' as in the example above)
- Putting an 'm2' seems to also work, that however is something we are explicitly told not to do:

+
[image]

https://linuxcnc.org/docs/devel/html/remap/remap.html#cha:remap

rmu75, 2023-11-22:
> It seems that an error during gcode execution will reset the stat.gcodes to G54 but not parameter[5220].

How did you check the parameter?

I think the "current" WCS is somewhat poorly defined in the interpreter. For reasons of path optimization, the interpreter is running way ahead of where the machine is, so the GUI has to look at the state tags of the motion currently executing. If there is no motion happening then it's (probably) the EMC_STAT where info comes from. Maybe in this error case some event that DRO / backplot needs updating is lost (in the GUIs), but it may also be possible something else is at fault.

Sigma1912, 2023-11-22:
> How did you check the parameter?

Basically what I found is:

sometimes 'self.stat.gcodes' disagrees with 'self.status.g5x_index'
and it seems 'self.status.g5x_index' reflects parameter[5220]
while 'self.stat.gcodes' gets updated in some other way.

When this happens the position values in the DRO do not reflect the internal state anymore.

Background:

The reason I came across this behavior is that I wanted to use emccanon. commands in a python remap to change the current WCS like this:
```

                self.params[5220] = 9
                emccanon.SET_G5X_OFFSET(
                    9,
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
                )
                yield INTERP_EXECUTE_FINISH
```

What I noticed is that  in AxisGUI the DRO will change the WCS to 'G59.3 ' (ie 9) however the Active Gcodes (ie stat.gcodes) is not updated to the new WCS (ie it does not change to 'G59.3'). 

In Gmoccapy the WCS index in the DRO follows the one in 'Active GCodes' so I  investigated this line 247  in 'combi_dro.py' (used in gmoccapy):

        self.gstat.connect('user-system-changed', self._user_system_changed)

which is indeed executed when the above emccanon. command runs and eventually executes this code that uses status.gcodes to display the current WCS:

```
    # get the actual coordinate system to display it on the DRO
    def _get_current_system(self):
            gcode = self.status.gcodes[1:]
            for code in gcode:
                if code >= 540 and code <= 590:
                    return "G%s" % int((code / 10))
                elif code > 590 and code <= 593:
                    return "G%s" % int((code / 10.0))
            return "Rel" 
```

So basically 'Gstat' correctly sends the message that there has been a change in the WCS system but because this change is not reflected in self.status.gcodes the DRO does not change.

Note that if we use 'status.g5x_index' in the above code:

           code = self.status.g5x_index

 the Gmoccapy DRO will then also disagree with the 'Active Gcodes' which leads me to presume that 'status.g5x' reflects the state of parameter[5220].

Only later did I realize that this state can also occur when not using python remaps and when this occurs  that the position in the DRO in both Axis and Gmoccapy and likely all the other Guis is also wrong.

Unfortunately I'm usually quickly lost as soon as C/C++ code is involved which seems to be where  the 'status' channel is coming from.

rmu75, 2023-11-22:
IIRC  'self.stat.gcodes' comes from motion state tags, and that is only valid while some sort of motion is happening.

The whole code around that is kinda messy and has probably bugs. On the gui side some unification is needed, now there are too many parallel implementations of basic logic which are (probably) slightly different.

Sigma1912, 2024-03-02:
Looking at this, it seems to me that parameter values may only get updated through gcodes (and m2?):
![parameters_set_by_gcode](https://github.com/LinuxCNC/linuxcnc/assets/
46067220/7e83d3fc-9b26-4f0f-8023-fdae1b7fb275)

Is this correct?
Which would maybe explain why a reset on abort would leave the parameter values out of sync with the system

On the other hand we have two 'SET_G5x_OFFSET' functions:
[image]

Is one of these where the 'self.stat' tags come from?
If so what is the other?

Sigma1912, 2024-03-13:
> If there is no motion happening then it's (probably) the EMC_STAT where info comes from.

Not sure if that is what you meant by 'EMC_STAT' but it's not coming from 'emcStatus->task.g92_offset.tran.x' (I can change that value in halui.cc and the DRO does not change it's behavior at all)

Sigma1912, 2025-06-06:
Seems the underlying issue is that the interpreter is running this part of the code in 'emc/rs274ngc/interp_convert.cc' and thus resets the parameters but due to the abort the corresponding 'PROGRAM_STOP()' it adds to the command queue is never actually executed.

[image]

Removing the highlighted code above makes the bug disappear (but it will then also not do a reset on a 'm2' that is actually executed)