LinuxCNC/linuxcnc #2748: Pncconf stepgen spindle doesn't ramp spindle speed

pcw-mesa, 2023-11-15:
When a stepgen driven spindle is setup with pncconf
the stepgen enable pin is connected to spindle on.
 
This cause instant steps in the spindle speed when the spindle
is turned off or a speed of 0 is set.

It would be better to connect the stepgen enable to machine on
allowing use of the stepgen acceleration limits to control the spindle
accel/decel 

 That is replace

net spindle-enable          =>  [HMOT](CARD0).stepgen.0n.enable
 with
net  machine-is-on           =>  [HMOT](CARD0).stepgen.0n.enable

c-morley, 2023-12-01:
Doesn't  the PID enable pin need to be switched too?

pcw-mesa, 2023-12-05:
The PID enable  can remain being controlled by spindle enable, 
as it will not cause an instant stop from the stepgen when its velocity 
command output is set to 0.

c-morley, 2023-12-09:
Thank you. Pushed to 2.9

closed 2023-12-09 by c-morley