LinuxCNC/linuxcnc #2752: RPi5 and hm2_rpspi (I know they are new to market and can be hard to get)

ozzyrob, 2023-11-18:
Some initial info here:
https://forum.linuxcnc.org/18-computer/50697-rpi5-spi-woes#285681

First thing of note in the driver the second 32 bit value of the RPi5 /proc/device-tree/soc/ranges is 0x10 compared to 0x00 for the RPi4.
Hexdump: 7c 00 00 00 00 00 00 10   7c 00 00 00 04 00 00 00
 
RPi5 with a 7i90 (hostmot firmware running on a Spartan 6 dev board, works fine with my RP-400)

 

Linuxcnc 2.9 recent pull from git hub
Initial play with the code was as such from RIP

	if(!pmembase || pmembase ==0x10) {
		// This is (probably) a RPi4 and the ranges file has a zero at
		// the base address. Here we need to read four 32-bit words to
		// get to the right values.
		rtapi_kfree(buf);
		if(!(buf = read_file("/proc/device-tree/soc/ranges", 4095, 16))) {
			rtapi_print_msg(RPSPI_ERR, "hm2_rpspi: Cannot determine IO base address and size.\n");
			return -1;
		}
		pmembase = be32toh(((uint32_t *)buf)[2]);
		pmemsize = be32toh(((uint32_t *)buf)[3]);
	}

got a result as such:
rmmod: ERROR: Module spi_bcm2835 is not currently loaded
hm2_rpspi: ERROR: Failed to execute '/sbin/rmmod spi_bcm2835'
hm2_rpspi: Platform: Raspberry Pi 5 Model B Rev 1.0
hm2_rpspi: Base address 7c000000 size 04000000
hm2_rpspi: Mapped peripherals from 0x7c000000 (size 0x04000000) to gpio:0x0x7fff1c1d0000, spi:0x0x7fff1c1d4000, aux:0x0x7fff1c1e5000
hm2_rpspi: No VPU clock at '/sys/kernel/debug/clk/vpu/clk_rate' (errno=2), trying core clock as alternative.
hm2_rpspi: SPI0/CE1 clock rate: 31250000/31250000 Hz, VPU clock rate: 50000000 Hz
hm2_rpspi: SPI0/CE1 write clock rate calculated: 25000000 Hz (clkdiv=2)
hm2_rpspi: SPI0/CE1 read clock rate calculated: 25000000 Hz (clkdiv=2)
Then just waiting for component to become ready.

With the unmodified driver from recent deb package.

hm2_rpspi: Platform: Raspberry Pi 5 Model B Rev 1.0
hm2_rpspi: Base address 0x00000010 size 0x7c000000
hm2_rpspi: Can't map peripherals
hm2_rpspi: cannot map peripheral memory.
hm2_rpspi: rtapi_app_main: Invalid argument (-22)
<stdin>:2: waitpid failed /usr/bin/rtapi_app hm2_rpspi
<stdin>:2: /usr/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for hm2_rpspi failed, returned -1

Dir listing of /proc/device-tree/soc if of any use

/proc/device-tree/soc $ ls
#address-cells'
avs-monitor@7d542000
clock@7c700000
compatible
cprman@7d202000
dma-ranges
fb
firmware
firmwarekms@7d503000
fixedregulator_3v3
fixedregulator_5v0
gpio@7d508500
gpio@7d517c00
gpiomem@7d504100
gpiomem@7d508500
gpiomem@7d510700
gpiomem@7d517c00
hdmi@7ef00700
hdmi@7ef05700
i2c@7d005000
i2c@7d005600
i2c@7d005800
i2c@7d005a00
i2c@7d005c00
i2c@7d005e00
i2c@7d508200
i2c@7d508280
i2c@7d508300
i2c@7d517a00
i2c@7d544000
_i2s@7d003000
intc@7d503000
intc@7d508380
intc@7d508400
intc@7d517000
intc@7d517ac0
intc@7d517b00
interrupt-controller@7c502000
interrupt-controller@7d510600
local_intc@7cd00000
mailbox@7c013880
mmc@7d002000
mop@7c500000
moplet@7c501000
name
phandle
pinctrl@7d504100
pinctrl@7d510700
pixelvalve@7c410000
pixelvalve@7c411000
power
pwm@7d00c000
pwm@7d00c800
pwm@7d517a80
ranges
rng@7d208000
rpi_rtc
serial@7d001000
serial@7d001400
serial@7d001600
serial@7d001800
serial@7d001a00
serial@7d50c000
serial@7d50d000
serial@7d50e000
'#size-cells'
sound
spi@7d004000
spi@7d004600
spi@7d004800
spi@7d004a00
spi@7d004c00
timer@7c003000
usb@7c480000
watchdog@7d200000

ozzyrob, 2023-11-18:
Few changes to src/hal/drivers/mesa-hostmot2/spi_common_rpspi.h 

#define BCM2835_GPIO_OFFSET	0x200000
#define BCM2835_SPI_OFFSET	0x204000
#define BCM2835_AUX_OFFSET	0x215000
*/

#define BCM2835_GPIO_OFFSET	0x1508500
#define BCM2835_SPI_OFFSET	0x1004000
#define BCM2835_AUX_OFFSET	0x1504100

Yielded this result
alcmd: loadrt hm2_rpspi spi_probe=2 spi_debug=4
hm2_rpspi: Platform: Raspberry Pi 5 Model B Rev 1.0
hm2_rpspi: Base address 7c000000 size 04000000
hm2_rpspi: Mapped peripherals from 0x7c000000 (size 0x04000000) to gpio:0x0x7fff894d8500, spi:0x0x7fff88fd4000, aux:0x0x7fff894d4100
hm2_rpspi: No VPU clock at '/sys/kernel/debug/clk/vpu/clk_rate' (errno=2), trying core clock as alternative.
hm2_rpspi: SPI0/CE1 clock rate: 31250000/31250000 Hz, VPU clock rate: 50000000 Hz
hm2_rpspi: SPI0/CE1 write clock rate calculated: 25000000 Hz (clkdiv=2)
hm2_rpspi: SPI0/CE1 read clock rate calculated: 25000000 Hz (clkdiv=2)
hm2_rpspi: SPI0/CE1 Invalid cookie, read: 00000000 00000000 00000000, expected: 55aacafe 54534f48 32544f4d
hm2_rpspi: SPI0/CE1 No drive seen on MISO line (kept at pull-down level). No board connected or bad connection?
hm2_rpspi: rtapi_app_main: No such device (-19)
<stdin>:2: waitpid failed /home/cnc/linuxcnc-dev/bin/rtapi_app hm2_rpspi
<stdin>:2: /home/cnc/linuxcnc-dev/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for hm2_rpspi failed, returned -1

I can do a successful mesaflash readhmid

cakeslob, 2025-02-09:
the new driver got merged, does anyone want to close this?

andypugh, 2025-02-09:
We probably also need to point out in the docs that hm2_rpspi doesn't work with Pi5?

cakeslob, 2025-02-09:
I think BAS updated all the docs, but perhaps he could provide his input on the matter, because its unclear to me if hm2_rpspi was made redundant by the hm2_spix driver, 

but it looks like the doc page for hm2_rpispi needs to be updated for rpi4 as well

BsAtHome, 2025-02-10:
Both the hm2_rpspi and the hm2_spi drivers are made redundant by the hm2_spix driver and the docs should be updated to have them point to the hm2_spix driver.

There is one tiny snag in the <i>html</i> documentation of the hm2_spix driver (adoc and man-format page are fine). The html tables are not shown, which is a real problem because it causes you to miss important information. Not sure yet how that can be fixed permanently.

andypugh, 2025-03-03:
Docs updated to deprecate the other spi drivers. 

2eccf7d21bd671fa7756f7937b73f92ad7717701

closed 2025-03-03 by andypugh