LinuxCNC/linuxcnc #2755: runtests with pyvcp

tjtr33, 2023-11-20:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1. in rip, i  create tesst/pyvcp.  and created  the test.hal lvl.xml and expected ( see attachments)
 2. in root of rip  run  scripts/runtests tests/[yvcp  and get erro "no display name and no $DISPLAY 
 3. cd to tests/pyvcp and run halrun -v -f tets.hal and get no error

## This is what I expected to happen:
I expect the same result from either test, 
as docs say the runtests use 'halrun -f tets.hal'

I donmt know what to put inn 'expected' untill success

## This is what happened instead:
scripts/runtests tess/pythin  reports failure wuth error "no display name and no $DISPLAY 
while printenv and 'echo$DISPLAY' report :0.0

## It worked properly before this:
I never tried scripts/runtests before,

(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:
thaitowerd10                
    description: Desktop Computer
    width: 64 bits
    capabilities: smbios-2.4 dmi-2.4 smp vsyscall32
   ...
       description: Motherboard
       product: DG41RQ
       vendor: Intel Corporation
...
*-cpu
          description: CPU
          product: Pentium(R) Dual-Core  CPU      E6300  @ 2.80GHz
...
*-memory
          description: System Memory
          physical id: 2a
          slot: System board or motherboard
          size: 4GiB
...
*-display
                description: VGA compatible controller
                product: Oland [Radeon HD 8570 / R7 240/340 OEM]
                vendor: Advanced Micro Devices, Inc. [AMD/ATI]
                physical id: 0
                bus info: pci@0000:01:00.0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: pm pciexpress msi vga_controller bus_master cap_list rom
                configuration: driver=radeon latency=0
                resources: irq:26 memory:c0000000-cfffffff memory:d0000000-d003ffff ioport:e000(size=256) memory:c0000-dffff

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * tomp@ThaiTowerD10:~/linuxcnc-barwidgets$ lsb_release -a
 * 
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

 * I am using this kernel version (shown by `uname -a`):
 * 
 * tomp@ThaiTowerD10:~/linuxcnc-barwidgets$ uname -a
Linux ThaiTowerD10 4.19.0-22-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.260-1 (2022-09-29) x86_64 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X ] A binary I built myself
   *      running linuxcnc from the rip reports  LINUXCNC - 2.10.0~pre0
   *     the next questiuon lolks like same question so not sure 
   *    but i built this rip (made the binary linuxcnc) so i imagine same  answer is expected
   * [ ] A binary version from some other source besides linuxcnc.org
   * [ ] 

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 *          LINUXCNC - 2.10.0~pre0

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 *    simply the xfce3 desktop, so Xwindows i suppose, i do not use linuxcnc in this effort
 *    ??? is the DISPLAY err or talking about the .ini file DISPLAY??? there is no .ini file !
 
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
 *    there is no hardwate IO involved, only video out to tkinter window on XFCE4 desktop
[lbl.xml.txt](https://github.com/LinuxCNC/linuxcnc/files/13406525/lbl.xml.txt)
[test.hal.txt](https://github.com/LinuxCNC/linuxcnc/files/13406526/test.hal.txt)

I attach 2 files renamed to ////txt because pf restrictions here
Ihe rop[ directory structure neeed to replicate errors is
( set the rip env up first, then  )
yourripdir/tests/pyvcp
    |--------- test/hal
     ----------lbl.xml
to test, cd into yourripdir/tests/pyvcp
       run    halrun -v -f test.hal      get no error, just a blink, use -I to examine the output longer than blink
then
cd to yourripdir and run   scripts/runtests tests/pyvcp
[lbl.xml.txt](https://github.com/LinuxCNC/linuxcnc/files/13406675/lbl.xml.txt)
[test.hal.txt](https://github.com/LinuxCNC/linuxcnc/files/13406676/test.hal.txt)

the is no visible cue that the attachments were attached
( usually seen at bottom of message)
and  i add them inline

#test.hal
###########################################
# 20nob test file lbl.hal
# #########################################
loadusr -Wn fred pyvcp -c fred lbl.xml
start

#lbl.xml
<?xml version='1.0' encoding='UTF-8'?>
<pyvcp>
	<vbox>
		 <label>
                	<text>"I am Groot"</text>
		</label>
	</vbox>
</pyvcp>

thanks tomp

tjtr33, 2023-11-20:
the attachments are visible in the text as hyperlinks  _after_  submitting

rmu75, 2023-11-21:
If you look at https://github.com/LinuxCNC/linuxcnc/blob/master/scripts/runtests.in#L7 this is done deliberately. `runtests` has to be able to run headless without requiring any user interaction.

tjtr33, 2023-11-21:
[rmu75](https://github.com/rmu75)
[Robert Schöftner](https://github.com/rmu75)
Showed this is expected behaviour. Closing as solved. Thanks Robert

closed 2023-11-21 by tjtr33