LinuxCNC/linuxcnc #2764: using M70/72 can unexpectaly start the spindle on abort or error.

c-morley, 2023-11-28:
If a user starts the spindle in MDI then stops the spindle with manual buttons, then uses a subprogram (such as probing routines) that record/reset codes with M70/M72 The spindle will start on abort or error.

I assume manual control doesn't reset the M5 in the internal settings.
 Explicitly setting M5 just before M70 will work around the problem. 

https://forum.linuxcnc.org/qtvcp/50729-qtdragonhd-versa-probe-failed-probe-turns-on-spindle?start=10#286614

complaint was on 2.9. master is the same - never tested 2.8.

rmu75, 2023-11-28:
"spindle stop with manual buttons" just sends a stop command to emctask, right? state in interpreter is not changed? so maybe the button should instead send "M5"?

rmu75, 2023-11-28:
btw, we really need a "max allowed rpm" field in the tool table, shouldn't be too hard to implement

c-morley, 2023-11-28:
In reality we must them in the GUI

c-morley, 2023-11-28:
Sorry missed a message....

The problem is manual stop and m5 are not equivalent in the internal settings.

I'm not sure why we still have manual/MDI modes in linuxcncs motion controller. We mix the modes in the GUI all the time.

rmu75, 2023-11-28:
Main difference seems to be teleop/free mode in manual vs. coordinated in mdi.

rmu75, 2023-11-28:
The problem with M70/M72 seems to be that interp internal "struct setup" is not in sync with status in emcmot. Don't know how to solve that -- why not make "external buttons" send "M5"?

c-morley, 2023-11-28:
Because you can't send M5 in Manual mode, only in MDI. Also the fix needs to be deeper then the GUI code (if that is what you mean) otherwise the problem is still lurking.

rmu75, 2023-11-28:
the problem is that emctask receives a NML command from the GUI to stop the spindle and instructs emcmot to do so. the interpreter has a separate status and doesn't see that command, so if it is told to restore state, in on_abort, it will restore to it's view of the world which has the spindle running.

either the gui button has to go through the interpreter or it has to reset the interpreter.

rmu75, 2023-11-28:
maybe make the button send M5 if in MDI mode and NML message if in manual? but that sounds like a hack and would need consistent impl. in every GUI. not a good solution.

c-morley, 2023-12-08:
Some probes require spinning spindles.
M70 is supposed to record the current state and it does not.
Maybe a better way to say this is the interpreter is not up to date with  linuxcnc's actual state. 

is there not a sync state command that MDI commands should call first or manual commands call after?

samcoinc, 2023-12-08:
wouldn't any que buster cause that?  Like a G4?

andypugh, 2023-12-09:
This seems like a difficult thing to solve in the general case. Even my idea above of writing to the interpreter state from motion won't always work if you consider pluggable interpreters. 

I think that this should be documented in the M70 documentation as a limitation of the code. 

I noticed yesterday that Chris has added an M5 to the script, and I hope to push out a 2.9 update in the next few days to distribute both this and the run-from-line fix. However that is dependent on working out how to fix the buildbot builds.

new952036, 2024-03-23:
@aman952036

new952036, 2024-03-23:
@aman952036