LinuxCNC/linuxcnc #2771: 'Run from here' not working as expected in Axis on 2.9.1

spacestate1, 2023-12-06:
## Here are the steps I follow to reproduce the issue:

 1. Open a gcode file in Axis
 2. Get XYZ homed 
 3. Scroll through the gcode in the code window to find an arbitrary point, right click and select 'Run from here' 

## This is what I expected to happen:

The machine re-positions the spindle and runs from the selected line of gcode 

## This is what happened instead:

The code runs from the beginning of the gcode file

## It worked properly before this:

I noticed the behavior start when I switched from  LinuxCNC 2.8.4 to 2.9.1
The same behavior is seen in 2.9.1 Axis_mm in sim running inside a virtual machine
   
## Information about my hardware and software:
 
 * I am using this user interface AXIS
 * I am using this interface hardware MESA 7i76 with a 7i92 interface card.
 * The issue is also present in sim Axis_mm

samcoinc, 2023-12-06:
Are you selecting the line before 'right clicking'?  If not - I think it
starts the program from the beginning..

sam

spacestate1, 2023-12-06:
I see the same behavior if I select the line and right click 'Run from here' or if I just right click and hit 'Run from here' on the line. it goes back to the beginning.

phillc54, 2023-12-06:
It appears to be caused by the file being reloaded and having the program_start_line reset from commit [bc9a79d](https://github.com/LinuxCNC/linuxcnc/commit/5c9a79d186ea5e8b64772c71694debaddb0005bb)

Sigma1912, 2023-12-07:
Yes, if iI recall correctly, the idea there was for the code to be reloaded when the user clicks on the new 'Re-Check' button on the limit violations popup and not force a reload when running the file.
```
def run_warn():
    warnings = []
    if o.canon:
        machine_limit_min, machine_limit_max = soft_limits()
        for i in range(3): # Does not enforce angle limits
            if not(s.axis_mask & (1<<i)): continue
            if o.canon.min_extents_notool[i] < machine_limit_min[i]:
                warnings.append(_("Program exceeds machine minimum on axis %s")
                    % "XYZABCUVW"[i])
            if o.canon.max_extents_notool[i] > machine_limit_max[i]:
                warnings.append(_("Program exceeds machine maximum on axis %s")
                    % "XYZABCUVW"[i])
    if warnings:
        text = "\n".join(warnings)
        return int(root_window.tk.call("nf_dialog", ".error",
            _("Program exceeds machine limits"),
            text,
            "warning",
            1, _("Re-Check"), _("Run Anyway"), _("Cancel"))) + 1
    return 0
```

andypugh, 2023-12-07:
Just to clarify, despite what Github says, I did not author the referenced change. It just merged PR 

> It appears to be caused by the file being reloaded and having the program_start_line reset from commit [bc9a79d](https://github.com/LinuxCNC/linuxcnc/commit/5c9a79d186ea5e8b64772c71694debaddb0005bb)

That change should only reload the file if the "Re-check" buttton is pressed. Is there something wrong with my logic?

phillc54, 2023-12-07:
TBH I did not notice a recheck button. I ran the axis_mm sim and just did a run from line and the file reloaded.

andypugh, 2023-12-07:
The change referenced ( bc9a79d )  should only be activated in cases where the limit check fails, and then the re-check button is pressed. 

Is it possible for someone to run a git bisect to see when the behaviour changed? (or just choose a point in the timeline prior to my change, to make sure it's my fault)

phillc54, 2023-12-07:
I can have a look tomorrow morning if no one else has by then. 
I did comment out the reload_file() line in the while res block and run from line worked correctly.

andypugh, 2023-12-07:
And this, kids, is why I shouldn't be allowed anywhere near software. 

Bother! It's an easy fix but I think it means a 2.9.2 release, and that's rather more work than the fix.

spacestate1, 2023-12-07:
add "if res == 0: break"  and it seems to fix the issue on my end.

closed 2023-12-08 by andypugh