LinuxCNC/linuxcnc #2773: Typo in Motion.c

TurboStreetCar, 2023-12-09:
Hello, it seems there’s a typo in the names_misc_errors array definitions inside of motion.c. 

It appears it’s using the count of array elements of names_dout instead of names_misc_errors, which effectively prevents you from using the names_misc_errors option. 

Line number 348-351. 

else if(names_misc_errors[0]){
    num_misc_error = count_names(names_dout);
    num_misc_error = (num_misc_error > count_names(names_din)) ? num_misc_error : count_names(names_din);
  }

ArchFabricationAndMachine, 2023-12-11:
Ive submitted a pull request on this issue.

hansu, 2023-12-12:
Please link to the corresponding pull request next time: https://github.com/LinuxCNC/linuxcnc/pull/2780

andypugh, 2023-12-15:
Fixed via https://github.com/LinuxCNC/linuxcnc/commit/74288982e684e03b567d9850efdd105dc209b1f4

closed 2023-12-15 by andypugh