LinuxCNC/linuxcnc #2784: /src/emc/usr_intf/emcrsh.cc function getToolOffset. Wrong param in snprintf.

johnchikov, 2023-12-11:
I think in emcrsh.cc (line 1677) use wrong argument type in snprintf instead of %f is %d

smoe, 2023-12-12:
This refers to https://github.com/LinuxCNC/linuxcnc/blob/7d9b5e504c1121f6d0be36120cd03a75ebf206da/src/emc/usr_intf/emcrsh.cc#L1677 and the question what the type emcStatus->task.toolOffset.tran.z may possibly be. That should be a double, right? And then, indeed, a "%lf" should possibly substitute the "%d".

If that is truly an error then this is also in version [2.9](https://github.com/LinuxCNC/linuxcnc/blob/fb4d6a175fcbfd279c069b2041173d90040a3f69/src/emc/usr_intf/emcrsh.cc#L1716), [2.8](https://github.com/LinuxCNC/linuxcnc/blob/fe43ab453bfd9728dbe32375403f75b53f7ce027/src/emc/usr_intf/emcrsh.cc#L1716), [2.7](https://github.com/LinuxCNC/linuxcnc/blob/d7ded6ef5e4c390267276dd72a59bfc4e319b5ce/src/emc/usr_intf/emcrsh.cc#L1602) and earlier, so one could assume.

smoe, 2023-12-13:
I just checked. It is a double since the ToolOffset is an [EmcPose](https://github.com/LinuxCNC/linuxcnc/blob/7d9b5e504c1121f6d0be36120cd03a75ebf206da/src/emc/nml_intf/emcpos.h#L19) , trans is a [PmCartesian](https://github.com/LinuxCNC/linuxcnc/blob/7d9b5e504c1121f6d0be36120cd03a75ebf206da/src/libnml/posemath/posemath.h#L121) of which the coordinates are doubles.

rmu75, 2023-12-13:
%f (or %g / %e) is the correct specifier for a double in printf, so as it is now it probably prints some garbled value. Even if repaired that method is somewhat incomplete as there may very well be additional offsets (besides Z).

smoe, 2023-12-15:
Since the value was garbled, the function cannot have been used much and, we have some freedom to correct it. What do you suggest? Add Z to the function name - and add variants for X and Y?

rmu75, 2023-12-15:
This is a can of worms. Not sure if we want to open it. in emcrsh after a quick glance, WCS offset commands return X, Y, Z and A, B, C offsets, linuxcnc knows of additional U, V, W offsets (linear, like X, Y, Z). I'm not sure what their purpose is (maybe in hot wire foam cutting?), but to be complete, all these values should be reported and probably also available as tool offset. Maybe make it depend on axis_mask.

Changing that would also affect / need changing the "set" commands in emcrsh.

heeplr, 2023-12-15:
@johnchikov can you retry with the current master? I did some [cleanup in linuxcncrsh](https://github.com/LinuxCNC/linuxcnc/pull/2739) that has been merged yesterday. 
I fixed this issue among others (and I hope I didn't introduce new ones) :)

Also, if you actually use linuxcncrsh, it'd be cool if you could do some real-world testing. 
Maybe we could bake that into a test with simulated hardware. Currently the tests for linuxcncrsh are not complete.

rmu75, 2023-12-15:
I'd say with 2.9 we only fix the %d and think about what offsets to display/read in 2.10