LinuxCNC/linuxcnc #2800: The world is not all Debian:     install with "sudo apt-get install bwidget"

garyemiller, 2023-12-20:
I'm trying to build linuxcnc from "git clone" on Gentoo.

I get things like this:

    install with "sudo apt-get install bwidget"

All the world is not Debian...

andypugh, 2023-12-21:
Is this any help? https://openlunchbox.com/mw19/index.php/HOWTO:_Gentoo_for_LinuxCNC

We only really attempt to support Debian and debian-like distros. We simply don't have the resources to support everything.

garyemiller, 2023-12-21:
FYI, here is how to install on a Gentoo host:

````
git clone https://github.com/LinuxCNC/linuxcnc.git
cd linuxcnc

euse -E tk
emerge -uDNa world

 python -m venv /path/to/venv
 . /path/to/venv/bin/activate

 pip install yapps2
 pip install gi

 emerge -a dev-libs/libmodbus
 emerge -a dev-lang/tk
 emerge -a dev-tcltk/bwidget
 emerge -a dev-tcltk/tclx

cd src
./autogen.sh
./configure --enable-non-distributable=yes
make
make setuid
````

I'm sure this not perfect, but enough for any Gentoo-er to get going.

smoe, 2023-12-25:
Is there a way to win you over to maintain a section in our documentation on how to build (and use) LinuxCNC on Gentoo?
Merry Xmas btw!

garyemiller, 2023-12-25:
> Is there a way to win you over to maintain a section in our documentation on how to build (and use) 
LinuxCNC on Gentoo? 

I've not figure dout how to use, and I assume that is the same for all users.  But I'd be happy to contribute a "Building for Gentoo" section.  What file should I look at?  I can submit an MR.

> Merry Xmas btw!

And a Happy New Year!

smoe, 2023-12-25:
May be something would fit into https://linuxcnc.org/docs/stable/html/code/building-linuxcnc.html ? The file to edit would be https://github.com/LinuxCNC/linuxcnc/blob/2.9/docs/src/code/building-linuxcnc.adoc .  Please note that I had selected the 2.9 branch for changes to the documentation that are likely compatible with the current stable version. The pull request should then also be against the 2.9 branch. Your contribution will then be merged with the master branch "a bit later".

garyemiller, 2023-12-25:
That looks like the right place.   Give me a few.days.

We need to all enjoy the season.

garyemiller, 2023-12-27:
Install insructions for gentoo:

[gentoo-diff.txt](https://github.com/LinuxCNC/linuxcnc/files/13781902/gentoo-diff.txt)

I forgot about this issue, so I also created a new issue for this.

petterreinholdtsen, 2023-12-27:
[Gary E. Miller]
> Install insructions for gentoo:

Perhaps you can create a pull request for this, with a 'closes'
reference to the relevant issues in the commit message?

-- 
Happy hacking
Petter Reinholdtsen