LinuxCNC/linuxcnc #2801: website not ADA copliant

garyemiller, 2023-12-21:
The website can be really hard to read.  It is clearly not
ADA compliant.

Take this page for example:

https://www.linuxcnc.org/documents/

The text at the bottom is small, and light gray font on a slightly lighter gray background.

andypugh, 2023-12-21:
I am not sure that the ADA applies to us, though that is no reason not to comply. 
However, to paraphrase the title of your other issue "The world is not all USA" We should probably look at WCAG instead. 
https://commonlook.com/international-web-accessibility-standards/

There is a list of tools here, from w3c, which can be used to check WCAG compliance. But bear in mind that LinuxCNC has (deliberately) a budget of $0 so any tool chosen would have to be FOS.

garyemiller, 2023-12-21:
ADA, for the web, in the USA is defined as WCAG, so we are on the same page  I know you don't need to comply, and EU demanding US follow its laws, and vice versa, is off topic here.  So lets just stick to the fact I'm going blind, but would like to read your web pages.  The WCAG is a really good place to start.

Small, and light gray font on a slightly lighter gray background. is really hard to read.  For some reason this became the fashion awhen COVID hit.  A simple CSS change will fix that.  If you can point me to the CSS, I'm happy to create a MR for you.

andypugh, 2023-12-21:
The web site is all here:
https://github.com/LinuxCNC/wlo

garyemiller, 2023-12-21:
As I expected, px, not pt.  Beginners mistake.  So the css ignores the DPI of the screen.  So if you view this on a high DPI screen, it gets real small:

````
  font-size: 16px;
````

Change all the px to pt, then adjust the sizes to your liking.  At 96 DPI:

1 Point = 0.75 * Pixel

But at 192 DPI:

1 Point = 1.5 * Pixel

The footer font is 9px, so 6pt type for me! 

And the font color is #828282 on #FDFDFD background.  Not a of of contrast given the small type gets aliased.  Maybe #525252 to give it some more contrast.

jethornton, 2023-12-21:
I never scroll down to the bottom of that page but I also don't like low contrast small type text as it's hard for me to read. I'll take a look at it in the morning (central time) and see if I can fix that for everyone.

garyemiller, 2023-12-21:
No rush for me.  A lot of beer is calling my name this weekend.

jethornton, 2023-12-21:
Hopefully it's a nice Weihenstephaner

JT

jethornton, 2023-12-22:
Everything I've read about web design say to not use pt for screens and only use it for printing.
Here is one page https://www.w3.org/Style/Examples/007/units.en.html

jethornton, 2023-12-22:
I've addressed the font size, links and contrast issues in the css file.

garyemiller, 2023-12-22:
> Everything I've read about web design say to not use pt for screens and only use it for printing.

Then we're reading conflicting opinions.  But in the end, I just want it readable.  And the new page is much better.  Thanks!

I'll go have a "Blonde Bombshell" to celebrate!  Followed by a double IPA.

closed 2023-12-22 by garyemiller