LinuxCNC/linuxcnc #2802: ./scripts/linuxcnc fonts really small

garyemiller, 2023-12-21:
I start a fresh install this way:

./scripts/linuxcnc

The text is really small.

How to make the text bigger?

andypugh, 2023-12-21:
In the config chooser? 
This appears to get a _little_ larger if the accessibility settings are changed to "large text" (there is also a "high contrast")
Though the main result of this is to make the window too small, requiring re-sizing, with a small and hard-to-target tool. 
There is also a screen-magnifier built into Linux.
I think that this issue is maybe better handled at the OS level rather than at the application level?

petterreinholdtsen, 2023-12-21:
[Gary E. Miller]
> The text is really small.
>
> How to make the text bigger?

What is the 'dimentions' and 'resolution' values under 'screen #0' in
the output from xdpyinfo when you run it locally?  I suspect they might
be related to your experienced font size.

These are the values on my machine, and here the fonts are ok.

  dimensions:    1366x768 pixels (361x203 millimeters)
  resolution:    96x96 dots per inch

-- 
Happy hacking
Petter Reinholdtsen

garyemiller, 2023-12-21:
Here is my xdpyinfo:

[xdpyinfo.txt](https://github.com/LinuxCNC/linuxcnc/files/13745743/xdpyinfo.txt)

I'm happy the font size is OK on your machine, how do I make it OK on mine?  My laptop has been running xfce for 5 years, and every other font is a good size for me.  Most apps allow the user to configure the fins sizes, or use the dsektop font sizes.

The common cause for odd font sizes is using pixels (px) instead of points (pt) in the CSS.  Points scale well, pixels don't scale at all.

garyemiller, 2023-12-21:
> In the config chooser?

I don't know what that means.  I did, and only did, this:

$ ./scripts/linuxcnc

And the text is tiny.

> accessibility settings are changed to "large text" 

Where is that?  If you mean the accessibility settings in xfce, I have already done that.

I use pixeltool, and the xfce ALT-[scroll] to view text.  But the x-height is so small, that zoomed in it is very few pixels to a character and hard to read.

andypugh, 2023-12-21:
> > In the config chooser?
> 
> I don't know what that means. I did, and only did, this:
> 
> $ ./scripts/linuxcnc

This isn't really relevant to the issue, but why? 
Are you building from source? (I guess so, if you are using Gentoo).  Ordinarily you would run . ./scripts/rip-environment and then just "linuxcnc" to start the program. 

Anyway, _where_ is the font tiny? Is it in the "LinuxCNC Configuration Chooser" or in the GUI once you open the software? And if the latter, which GUI are you using?

garyemiller, 2023-12-21:
> This isn't really relevant to the issue, but why?

Because the doc said "run linuxsrc" and that is where I found it.  Here is what I was reading:

http://linuxcnc.org/docs/stable/html/getting-started/running-linuxcnc.html

Is that wrong?

> Are you building from source? 

Do I have a choice on Gentoo?

> Ordinarily you would run . ./scripts/rip-environment

That's not what that web page says, and it fails:

````
gem@dilbert GRBL/linuxcnc $ scripts/rip-environment
This script can be used in one of two ways.

It can be loaded in the context of your shell, by executing
    . scripts/rip-environment
after this, commands like 'linuxcnc', 'halrun' and so on refer to the
version in this directory, instead of to an installed version of
linuxcnc (if any)

Second, it can be used to run a command in this directory without
modifying the environment of the calling shell:
    scripts/rip-environment command...
such as
    scripts/rip-environment halrun foo.hal
    scripts/rip-environment linuxcnc configs/.../foo.ini
````

> Anyway, where is the font tiny?

Screenshot:

[image]

I ideally I'd like the default font to be similar to, or slightly bigger, than the terminal font.  Most programs allow the user to adjust the font sizes.

garyemiller, 2023-12-21:
I managed to get the output of the "info" command:

[info.txt](https://github.com/LinuxCNC/linuxcnc/files/13746430/info.txt)

andypugh, 2023-12-21:
You missed the first . in the rip-environment command. its dot space dot slash

You can bypass the config picker altogether. (I think that most users do) by using a desktop shortcut that starts the config directly. There is an (admittedly tiny) checkbox at the bottom of the config picker to do this. 

For some reason the text on the config picker looks unusually small. 
I don't know where it is defined, but the config picker is created by this Tcl file:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/tcl/bin/pickconfig.tcl

petterreinholdtsen, 2023-12-21:
I see your xdpyinfo report this:

  dimensions:    1920x1080 pixels (508x285 millimeters)
  resolution:    96x96 dots per inch

Slightly more dots than mine.  Looking at the screen shots, I start to wonder if perhaps it is a question of opinion, as the fonts on your screen shots look OK to me, while you obviously find them too small.

garyemiller, 2023-12-21:
> You missed the first . in the rip-environment command. its dot space dot slash

Same results, either way:

````
gem@dilbert /mnt/backup/archive/GRBL/linuxcnc/scripts $ ./rip-environment 
This script can be used in one of two ways.

It can be loaded in the context of your shell, by executing
    . ./rip-environment
after this, commands like 'linuxcnc', 'halrun' and so on refer to the
version in this directory, instead of to an installed version of
linuxcnc (if any)

Second, it can be used to run a command in this directory without
modifying the environment of the calling shell:
    ./rip-environment command...
such as
    ./rip-environment halrun foo.hal
    ./rip-environment linuxcnc configs/.../foo.ini
````

> You can bypass the config picker altogether.

How do I do that?

> For some reason the text on the config picker looks unusually small.

Yes, that is my point, very small.

> Slightly more dots than mine. Looking at the screen shots, I start to wonder if perhaps it is a question of opinion, as the fonts on your screen shots look OK to me, while you obviously find them too small.

Different people will have different screen sizes, and so need to be able to scale for their hardware and their eyesight.  Somewhere something set the font size, I just want to change that.

> I start to wonder if perhaps it is a question of opinion, as the fonts on your screen shots look OK to me, while you obviously find them too small.

It also a matter that my screen  has more DPI than yours.  And my eyes are much worse than yours.

andypugh, 2023-12-21:
> > You missed the first . in the rip-environment command. its dot space dot slash
> 
> Same results, either way:

It might be "source" in Gentoo. 

>  > You can bypass the config picker altogether.
> 
> How do I do that?

As I said, tick the checkbox at the bottom of the config picker to create a desktop shortcut for the specific config.

garyemiller, 2023-12-22:
> As I said, tick the checkbox at the bottom of the config picker

Is there a web page that explains this?  So I don't have to me walk you through every step?