LinuxCNC/linuxcnc #2810: Add Gentoo build instructions.

garyemiller, 2023-12-27:
Update ./docs/src/code/building-linuxcnc.adoc with build instructions for gentoo:

[gentoo-diff.txt](https://github.com/LinuxCNC/linuxcnc/files/13781872/gentoo-diff.txt)

Author="Gary E. Miller" <gem@rellim.com>"

petterreinholdtsen, 2023-12-27:
See also #2800.

smoe, 2023-12-28:
@garyemiller , first and foremost have many thanks for caring about LinuxCNC and your patch to help promoting that technology on Gentoo. It is straight-forward to transform your patch into a pull request - it just takes some 5-15 minutes in which I could do something else, while accepting/rejecting a pull request would take just as long as it takes to read the text. And my "else" list is long - very long - too long - far too long - stressing me out. And me doing the work for you would drag in extra work for subsequent changes of yours to the text or (preferably) additional changes to the source tree.

If you would send some signals that you would be prepared to learn, would happily assisting you in that process (also per video if so required),  how to prepare proper pull requests then your change will likely find entry in the documentation. If not then, since outside the LinuxCNC core promise of where maintenance efforts should go, I do not see that happening.

Yes, you are right, me preparing the PR would likely have taken less time than me writing this comment :o)

garyemiller, 2023-12-28:
I'm used to GitLab, not GitHub.  I'm logged in to GitHub, but the "Create New Merge Request" is grayed out for me, and since I don't have privs on this repository I can't create a new branch.

So I don't know how to make an MR here.

I could use "git format-patch" to make a slightly better patch.  Then you just have to "git am" to apply it.

garyemiller, 2023-12-28:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

"Note: To open a pull request in a public repository, you must have write access to the head or the source branch or, for organization-owned repositories, you must be a member of the organization that owns the repository to open a pull request."

And I'm none of those...

phillc54, 2023-12-28:
I use the clone and push to my github repo method.
http://linuxcnc.org/docs/html/code/contributing-to-linuxcnc.html#_overview_of_the_process

garyemiller, 2023-12-29:
From that page:

"Share your changes with the other project developers in one of these ways:

    Push your branch to github and create a github pull request to https://github.com/linuxcnc/linuxcnc (this requires a github account), or"

Once again, I can't create a github pull request because I'm not on the project.

"Push your branch to a publicly visible git repo (such as github, or your own publicly-accessible server, etc) and share that location on the emc-developers mailing list, or"

I'd really rather not have to create a repo on service that will cost me money.

"Email your commits to the LinuxCNC-developers mailing list (<[emc-developers@lists.sourceforge.net](mailto:emc-developers@lists.sourceforge.net)>) (use git format-patch to create the patches)."

I just did that.

0001-building-linuxcnc.adoc-Add-gentoo-instructions.patch

smoe, 2023-12-29:
You already have all you need to create a pull request - a GitHub account that is. If you are still interested to get your instructions in - I had promised to help, here comes the help:

The process starts with that you clone the LinuxCNC repository, i.e. you go to https://github.com/LinuxCNC/linuxcnc/ and press the "Fork" button. This is like "GitHub runs 'git clone' on a machine of theirs and grants you (and you only) access to it".  If I recall correctly then you are directly guided towards that with your browser. You then add that second LinuxCNC repository, i.e. your clone of the original, as another remore repository to your current source tree that you had checked out from GitHub before. The command likely is:
 
 ```
git remote add myrepo git@github.com:garyemiller/linuxcnc.git
 ```
 You may want to run git status, checkout a new branch as in `git checkout -b gentoo` and commit the changes if you have not already with `git commit -m "docs: instructions to build on gentoo"`, and then push to your repository: `git push myrepo`. When opening https://github.com/LinuxCNC/linuxcnc/ (or equally on the page showing your repository) you will see a yellowish box guiding you prepare a pull request.
 
From how I now described the process, your changes will likely aim at the current development branch of LinuxCNC, not against 2.9, but as a start this may not be too bad.

Final comment - you may not be part of the GitHub organisation of LinuxCNC (I am not either)  - but as a user and very much so when you have your first patch to the documentation in - you are part of the LinuxCNC community. So, thank you again, and if you have ideas on how to futher strengthen the documentation such that a misunderstanding about GitHub can be avoided - please have a second PR coming.

garyemiller, 2024-01-03:
MR created.  Way too much work for a simple patch.

I guess someone can remove the instructions to email patches to the mailing list as that gets ignored...

smoe, 2024-01-03:
> MR created.

Thank you for https://github.com/LinuxCNC/linuxcnc/pull/2826 .

> Way too much work for a simple patch.

A PR is the only way to get anything in - for all of us. Why should anyone want to do it, outside of what is officially maintained if the main audience (the Gentoo community) does not want to invest the effort.

> I guess someone can remove the instructions to email patches to the mailing list as that gets ignored...

You have not been ignored. There is just nobody responsible for Gentoo besides you, so the patch would have been rejected if you did not perform the PR yourself ... and provide future updates. On a sidenote - that is where most time is sunk - maintenance.

Your next PR will be easier. Thank you again!

garyemiller, 2024-01-03:
I'll close this, deferring to the PR.

garyemiller, 2024-01-05:
I guess this needs to be open so the MR can refer to it.

andypugh, 2024-01-08:
I merged the PR (though the builds are currently failing, probably not due to the PR)

closed 2024-01-08 by andypugh