LinuxCNC/linuxcnc #2818: Gmoccapy - confirm value in Tool table

zz912, 2023-12-31:
RIP 2.9 Branche
Debian Bookworm

If I want save changed value in Tool table I must confirm changed value by picking mouse in tooltable or leave cell by Enter or by arrow.

This is a parody of what it looked like when I was learning to use the Tool table.
[image]

Could be possible confirm changed value by clicking on some horizontal button?

zz912, 2025-07-05:
If a person knows about this error, he will be careful about it.

@Sigma1912 is working with tooltable now. I wanted to ask if he could create a signal that would be triggered by the Apply button and would accept the value written in the cell. I tried that, but I couldn't.

zz912, 2025-07-06:
I guess we're misunderstanding each other. When you delete a tool, you don't have an unfinished cell.

This table is similar to:
- Microsoft Excel
- LibreOffice Calc
- Google Sheets
You dont have to confirm anything when you write something in Cell.

hansu, 2025-07-06:
But you have a save and an undo function there.
The apply button is equal to the save function.

What about this - move the apply button next to the exit button and perform two actions:
1. apply
2. return to main page
?

[image]

Sigma1912, 2025-07-06:
> You dont have to confirm anything when you write something in Cell.

That is true but in those applications you also have an undo button.

Sigma1912, 2025-07-06:
> What about this - move the apply button next to the exit button and perform two actions:

This does not change the behavior of unconfirmed edits being rejected.

I've been planning to implement moving the table editing buttons into the table frame as presented here:
https://github.com/LinuxCNC/linuxcnc/pull/3472#issuecomment-2988010896

In my opinion having unfinished edits rejected on leaving the page is not totally unexpected behavior.

hansu, 2025-07-06:
> > What about this - move the apply button next to the exit button and perform two actions:
> 
> This does not change the behavior of unconfirmed edits being rejected.

Yeah but you can just click on another cell for that input to be accepted. That is not uncommon. And only applies when you disable using the calculator for this.

>In my opinion having unfinished edits rejected on leaving the page is not totally unexpected behavior.

I agree

>I've been planning to implement moving the table editing buttons into the table frame as presented here:
https://github.com/LinuxCNC/linuxcnc/pull/3472#issuecomment-2988010896

You know, I am fine with this 😉

hansu, 2025-07-06:
Another thing that bothers me sometimes,  just throw it into the room here:

When changing a tool you need to deselect the current one and select the checkbox of the new one.
What if the tool of the selected row will be the next? So only clicking the change button will change from tool 5 to 10.
What do you think?

[image]

Sigma1912, 2025-07-06:
> When changing a tool you need to deselect the current one and select the checkbox of the new one.

Yes, that is a bit of a nuisance. 
However, it might  not be overly intuitive to have a 'Select' checkbox but use the highlighting for the tool to change to.

hansu, 2025-07-06:
Yeah that's a bit difficult - what should happen when one tool is selected by a checkbox and another one with the "line-select"? 🤔

closed 2026-02-09 by hansu