LinuxCNC/linuxcnc #2820: PNCconf - Gmoccapy need gmoccapy.tooloffset-x gmoccapy.tooloffset-z

zz912, 2023-12-31:
RIP 2.9 Branche
Debian Bookworm

I would like ask you for adding for Gmoccapy configuration

in main.hal:
```
net tooloffset-x  <=  motion.tooloffset.x
net tooloffset-z  <=  motion.tooloffset.z
```

in gmoccapy_postgui.hal:
```
net tooloffset-x  => gmoccapy.tooloffset-x
net tooloffset-z  => gmoccapy.tooloffset-z
```

This lines controls this:
[image]

I tried modify build_HAL.py , but it was difficult for me. There are many variants.

hansu, 2024-01-06:
May something like this: https://github.com/LinuxCNC/linuxcnc/pull/2838?

Not tested.

zz912, 2024-07-12:
PR https://github.com/LinuxCNC/linuxcnc/pull/2838 solved this Issue.

closed 2024-07-12 by zz912