LinuxCNC/linuxcnc #2825: module 'PIL.Image' has no attribute 'ANTIALIAS'

garyemiller, 2024-01-03:
Just built on Gentoo from git head:

````
$ bin/image-to-gcode /tmp/Screenshot_2024-01-02_12-38-54.png 
/home/gem/src/linuxcnc/bin/image-to-gcode:772: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  nim = numpy.fromstring(tobytes(im), dtype=numpy.uint8).reshape((h, w)).astype(numpy.float32)
Traceback (most recent call last):
  File "/home/gem/src/linuxcnc/bin/image-to-gcode", line 837, in <module>
    main()
  File "/home/gem/src/linuxcnc/bin/image-to-gcode", line 773, in main
    options = ui(im, nim, im_name)
              ^^^^^^^^^^^^^^^^^^^^
  File "/home/gem/src/linuxcnc/bin/image-to-gcode", line 522, in ui
    ui_image = im.resize((nw,nh), Image.ANTIALIAS)
                                  ^^^^^^^^^^^^^^^
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'
````

havardAasen, 2024-01-09:
If you are using master, the fix was included a couple of hours ago.

garyemiller, 2024-01-10:
Works now.  Thanks!

I do get this:

```` $ bin/image-to-gcode /tmp/Screenshot_2024-01-02_12-38-54.png 
/home/gem/src/linuxcnc/bin/image-to-gcode:772: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  nim = numpy.fromstring(tobytes(im), dtype=numpy.uint8).reshape((h, w)).astype(numpy.float32)
````

Should I file a new issue on that?

havardAasen, 2024-01-10:
> ```
> /home/gem/src/linuxcnc/bin/image-to-gcode:772: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
>   nim = [...]

No need, I created #2843 to fix that.

PR #2828 (2024-01-07): Update deprecated Pillow constant
The substituted constant, `ANTIALIAS`, was removed in Pillow 10.0.0. The `LANCZOS` constant was introduced in Pillow 2.7.0.
https://pillow.readthedocs.io/en/stable/releasenotes/10.0.0.html#constants

Fixes #2825
closed 2024-01-10 by garyemiller