LinuxCNC/linuxcnc #2839: Axis DRO not updating

cakeslob, 2024-01-06:
Hello, a user on the forum raised this issue, which I have known about, and experienced before. 

The Axis DRO tab does not update when the tab is raised. After the first time the DRO tab is raised and lowered, the DRO coordinated do not update unless commanded to. So a situation can arise when you move a joint in Preview tab, and switch to the DRO tab, the coordinates are not updated to reflect current position. 

Can anyone else verify this issue? I do not think is has been fixed. Im using an older linuxcnc, 2.9pre, but another user raised this issue so I will assume it is not fixed. 

I am deficient in computer hacking prowess, so I do not know exactly how to fix this. I think it will be something like this, but I dont know how to pass the command properly

`${pane_top}.right itemconfigure numbers -raisecmd "[redraw_dro];"` 

## Here are the steps I follow to reproduce the issue:

1. - Open preview, move, do what ever
2. - Open DRO, at this point they are synced
3. - Open Preview, move 5mm
4. - Open DRO, at this point the DRO should be 5mm behind the Preview
5. - Press Top view (or any view ) button from the toolbar, and watch the DRO fix itself

## This is what I expected to happen:
DRO tab to read the same position as preview DRO
## This is what happened instead:
DRO remains un-updated when the tab is raised, displaying potentially incorrect position

phillc54, 2024-01-06:
Could you try this patch to see if it fixes the issue.
[axis_dro_update.patch](https://github.com/LinuxCNC/linuxcnc/files/13851662/axis_dro_update.patch)

andypugh, 2024-01-08:
Do you need help with how to apply and test the patch?

Sigma1912, 2024-01-08:
Seems to have worked out:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/51242-axis-2-9-2-position-in-preview-dro-tabs-aren-t-in-sync-bug?start=0#290023

closed 2024-01-09 by phillc54