LinuxCNC/linuxcnc #2842: API for libgpiod has changed in v2

jasonKercher, 2024-01-09:
## Here are the steps I follow to reproduce the issue:

 1. Have libgpiod version >= 2
 2. Attempt to compile linuxcnc v 2.9.2

## This is what I expected to happen:

Build succeeds

## This is what happened instead:

```none
In file included from hal/drivers/hal_gpio.c:39:
hal/drivers/hal_gpio.c:188:50: error: invalid application of ‘sizeof’ to incomplete type ‘struct gpiod_line_bulk’
  188 |             (*ptr)[c].bulk = rtapi_kmalloc(sizeof(*(*ptr)[c].bulk), RTAPI_GFP_KERNEL);
```

...along with lots of warnings for implicit declarations of functions (due to API change).

## It worked properly before this:

libgpiod version <= 1.6.4

## Information about my hardware and software:

 * I am using Manjaro.  I know that is not officially supported, but we do keep a working [linuxcnc package](https://aur.archlinux.org/packages/linuxcnc) over there.
 Manjaro is part of the Arch Linux family, so we always have the bleeding edge latest software.  I imagine this will not be an issue on Debian for some time.
 * Linux Kernel Version 6.1.66
 * Linux CNC v2.9.2

andypugh, 2024-01-09:
This was something that I knew was coming, and that I don't really know how to solve. 

libgpiod is a mess, for example libgpiod2 is still v1.6, not 2. 

There are no packages in Debian  for libgpiod v2, with the new API, so I decided this was a problem that can be kicked down the road until at least after Trixie (which still has v1.6) https://packages.debian.org/search?suite=trixie&searchon=names&keywords=libgpiod

There is probably something horribly messy that can be done with conditional compilation, or a wrapper API. 

I am not even 100% sure that libgpiod has a 1:1 mapping between v1.x and v2.x functions. 

The libgpiod developers were keen that I should code for v2, but with no packages available from Debian, how would our users get it? We can host the packages on our own repository, but that isn't a fix for folk installing LinuxCNC from the main Debian mirrors.

FabioLolix, 2024-01-09:
Hello, on the AUR making a pkgbuild for libgpiod v1.6.4 installed in /opt/libgpiod-1.6.4 (or something similar) so it doesn't conflicts with the system's one isn't a problem

How I could pass that to autogen.sh or ./configure ?

FabioLolix, 2024-01-09:
btw [https://github.com/pikvm/ustreamer](ustreamer) updated to build with libgpiod 2.0.x while also working with the older version seems, the 2 relevant commits:

* https://github.com/pikvm/ustreamer/commit/d9b5f2b03d079396548f71af3c2036b8433c6961
* https://github.com/pikvm/ustreamer/commit/4d92dc662cf4cfa7a648cf5ef58d0ccc038f34c9

MHoyer12345678, 2024-11-21:
> This was something that I knew was coming, and that I don't really know how to solve.
>
> libgpiod is a mess, for example libgpiod2 is still v1.6, not 2.
>
> There are no packages in Debian for libgpiod [...]

MHoyer12345678, 2024-11-21:
> The libgpiod developers were keen that I should code for v2, but with no packages available from Debian, how would our users get it? We can host the packages on our own repository, but that isn't a fix for folk installing LinuxCNC from the main Debian mirrors.

I debianized the 2.2.x branch to work w/ the new api. Maybe this can help someone else as well when trying to use the new API ...

andypugh, 2025-12-26:
There is now a PR that should address this:
https://github.com/LinuxCNC/linuxcnc/pull/3678