LinuxCNC/linuxcnc #2869: Hostmot2

JTrantow, 2024-01-29:
# The issue tracker is not a support forum

#2860 was prematurely closed. The fix works for the halrun script that demonstrated the problem but the issue still exists when you run a  full configuration in linuxcnc.

## Here are the steps I follow to reproduce the issue:

 1. Load a mesa config with at least one encoder and the following connections
net machine-is-on      <= halui.machine.is-on => hm2_7i92.0.encoder.00.quad-error-enable 
net estop-x-encoder   <= hm2_7i92.0.encoder.00.quad-error => joint.0.amp-fault-in
 2. Create a quad error on the x encoder by toggling the input a/b simultaneously.
 3. Click the power button which sets the machine-is-on and enables the quad-error-enable.

## This is what I expected to happen:
Any quad errors that occur when the encoder quad-error-enable is false should be discarded.

## This is what happened instead:
Quad errors that occur when quad-error-enable is false immediately pop up when quad-error-enable is set true by machine-is-on.

pcw-mesa, 2024-01-31:
Should be fixed in 2.9 and master now

JTrantow, 2024-02-05:
Latest fix by PCW tested on my lathe and mill resolve this issue.

closed 2024-02-05 by JTrantow