LinuxCNC/linuxcnc #2878: Python Interface Returns Wrong State

jethornton, 2024-02-04:
task_state

    (returns integer) - current task state. one of STATE_ESTOP, STATE_ESTOP_RESET, STATE_ON, STATE_OFF.

STATE_OFF is never returned, when you  set the state(int) to STATE_OFF STATE_ESTOP_RESET is returned.

    Set the machine state. Machine state should be STATE_ESTOP, STATE_ESTOP_RESET, STATE_ON, or STATE_OFF.

Axis is running and I use the estop and power buttons in axis to change states

```
john@cave:~$ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import linuxcnc
>>> s = linuxcnc.stat()
>>> c = linuxcnc.command()
>>> linuxcnc.STATE_ESTOP
1
>>> linuxcnc.STATE_ESTOP_RESET
2
>>> linuxcnc.STATE_ON
4
>>> linuxcnc.STATE_OFF
3
>>> # estop on
>>> s.poll()
>>> s.task_state
1
>>> # estop off
>>> s.poll()
>>> s.task_state
2
>>> # power on
>>> s.poll()
>>> s.task_state
4
>>> # power off
>>> s.poll()
>>> s.task_state
2

```

jethornton, 2024-02-04:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/axis/extensions/emcmodule.cc#L419

Says         "    STATE_OFF: Same as STATE_ESTOP_RESET, this one is not used.\n"

So the Docs are wrong?

hydroid7, 2024-02-11:
`STATE_OFF` is never returned as state, since the state machine immediately jumps to `STATE_ESTOP_RESET`. But you can issue a `STATE_OFF` request to disable linuxcnc:
```python
c.state(STATE_OFF)
```
to get into `STATE_ESTOP_RESET` state.

jethornton, 2024-02-11:
So the docs are not correct stating that STATE_OFF is returned.