LinuxCNC/linuxcnc #2889: Stat.axes returns error

jethornton, 2024-02-15:
axes is in the documents http://linuxcnc.org/docs/stable/html/config/python-interface.html#_linuxcnc_stat_attributes

But does not exist.

john@cave:~$ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import linuxcnc
>>> s = linuxcnc.stat()
>>> s.poll()
>>> s.axes
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'linuxcnc.stat' object has no attribute 'axes'. Did you mean: 'axis'?

phillc54, 2024-02-15:
Looks like it was removed in 2.9 but not from the docs.

This is [2.8](https://github.com/LinuxCNC/linuxcnc/blob/334d81c5277982b5b73043d5cde7f119635de629/src/emc/usr_intf/axis/extensions/emcmodule.cc#L688)

Sigma1912, 2024-02-16:
A pull request is pending to fix the documentation:
https://github.com/LinuxCNC/linuxcnc/pull/2789

jethornton, 2024-02-16:
Seems the function should be fixed as there is no replacement for it.

Sigma1912, 2024-02-16:
> Seems the function should be fixed as there is no replacement for it.

We have already had this discussion on the pull request page, I don't understand why you keep insisting on putting the axes attribute back in the status channel when one line of python code will give you what you want:

number_of_axes = (s.axis_mask).bit_count()

jethornton, 2024-02-16:
Because who in the world would know how to do that? You sure, 99% of the rest of us it's more usable to just use s.axes. I don't understand why it was removed being such a nice thing to have. I also don't understand your opposition to having the axes stat...

Sigma1912, 2024-02-16:
The point is that all these attributes have to be sent  on every single poll of the status channel. It's simply inefficient to send the same information twice.

I'm not a programmer. There is no shame in not knowing all the functions python has to offer, I also did not know to do it until I tried to help you with your issue. It took me maybe 15 minutes of online research to figure out how to do it. I even went through the trouble of trying it out and posting my findings for you and others. Yet all I got is you ranting about wanting it the old way.  
I just find it surprising that somebody who is obviously clever enough to have used linuxcnc and it's python interface for years would spend that much time and energy to have redundant information put back in the code.
Clearly it would have taken less effort to create a pull request to update the  'python-interface.adoc' on how to use .bit_count to extract the information from the 'axis_mask' attribute.
And with this I rest my case.

closed 2024-02-21 by jethornton