LinuxCNC/linuxcnc #2891: Typo regarding "G52", "G92" and DISABLE_G92_PERSISTENCE

Unlogic-dot-se, 2024-02-18:
On the page [linked here](https://linuxcnc.org/docs/html/gcode/overview.html) it says:

- 5210 - 1 if "G52" or "G92" offset is currently applied, 0 otherwise. Volatile by default; persistent if DISABLE_G92_PERSISTENCE = 1 in the [RS274NGC] section of the INI file.

- 5211-5219 - Shared "G52" and "G92" offset for X, Y, Z, A, B, C, U, V & W. Volatile by default; persistent if DISABLE_G92_PERSISTENCE = 1 in the [RS274NGC] section of the INI file.

This seems to have been inverted, shouldn't it say that the parameters in question are persisted by default and volatile only if "DISABLE_G92_PERSISTENCE = 1" is set?

andypugh, 2024-02-27:
It probably should. 
But do you have the time to check what actually happens? Possibly the INI file entries have confusing names?

Sigma1912, 2024-03-04:
related https://github.com/LinuxCNC/linuxcnc/pull/2915

Sigma1912, 2024-03-06:
Oh my, 'DISABLE_G92_PERSISTENCE = 1' is actually broken.
 I had tested with my on_abort command (which also runs on 'machine on') which hid the issue.

I'll have a look.

hansu, 2024-09-16:
@Unl0gic Is this resolved now?

Unl0gic, 2024-09-20:
@hansu Yes from what I can see this issue has now been corrected.

closed 2024-09-20 by andypugh