LinuxCNC/linuxcnc #2898: Clean up Stale Branches

satiowadahc, 2024-02-19:
There are over 300 branches some which have not had development in 10-12 years... could we look into deleting 1 or 2 of them?

satiowadahc, 2024-02-19:
Adding to this, most projects I've been a part on, the git flow is make your own fork, make changes, do a PR from your fork back to the project.

andypugh, 2024-02-19:
Many probably can be deleted, though I regret deleting my own G71-remap branch because I have frequently wanted to reference parts of it since (either the documentation of the cycles, which was better than what is currently in the docs, or just specific examples of how to build a remap using canonical commands).
So, it's probably something for the branch "owners" to do, rather than a third party.

samcoinc, 2024-02-19:
I probably have this branch on a machine - do you want me to look for it?

satiowadahc, 2024-02-19:
I get wanting to keep branches, but its nicer if the main project doesn't have a ton of personal development branches. IE, just making a PR for 2.9 just now and I couldn't find the branch as there is so many.

jethornton, 2024-02-19:
Is there a way to archive the old personal branches?

A lot of personal branches were created by people that no longer participate in this project.

rmu75, 2024-02-19:
branches are not part of the history. the commits don't go away if you delete a branch, but they may get garbage collected if nothing references them any more. forking the repository into some branch-backup and removing the branches in the main repo should be safe. or set a tag on each branch tip before removing the branch, but that would pollute the tag namespace.

satiowadahc, 2024-02-19:
Looks like another option is to fork the repository, and push stale branches to a separate project.

Looking at my fork, all branches exist on it that were there at the time of my fork.

satiowadahc, 2024-02-19:
on that note @andypugh is this the branch you meant? https://github.com/satiowadahc/linuxcnc/tree/andypugh/g71type2remap

jethornton, 2024-02-21:
I "think" I deleted all my old stale branches...

satiowadahc, 2024-02-21:
5 down 342 to go...

andypugh, 2024-02-21:
I deleted 15 of mine (which have either been superceded or already merged)

andypugh, 2024-02-21:
Is there a way to tell if a feature branch was merged? For example
https://github.com/LinuxCNC/linuxcnc/commit/663119abb3a2eb7b9ed881420953fe83a494de63
I can't work out if this bit: 

[image]

Is indicating that the change went into 2.7...

satiowadahc, 2024-02-21:
It is in master... https://github.com/LinuxCNC/linuxcnc/commits/master/src/emc/nml_intf/emc.hh?after=ee58250f0a22e1657f849b271ee4ed0a7953f131+34

satiowadahc, 2024-02-21:
e9613c25347b76e9ac3a15841112a7d0b282a30b

I use gitkraken, and just scrolled back through history.

satiowadahc, 2024-02-21:
[image]

satiowadahc, 2024-02-21:
```bash                                                                       
#!/bin/bash

branches=($(git branch -r))
numbranches=${#branches[@]}

echo $numbranches

for (( c=0; c<340; c++ ))
do
   tag=$(git describe --tags ${branches[$c]})
   commit=$(echo $tag | sed 's/.*-g//')
   # echo "Branch ${branches[$c]} : Commit: $commit"
   itsthere=$(git branch master --contains $commit)
   if [ "${itsthere}" != "" ]
   then
      echo "Branch ${branches[$c]} : Commit $commit in master"
   fi
done
```

Output:

```bash
Branch linuxcnc/2.8-pocket-fix : Commit f2cfff8945a in master
Branch linuxcnc/Issue105 : Commit 5bdebef86dc in master
Branch linuxcnc/bad-merge-20200417 : Commit 2e4c10e05e9 in master
Branch linuxcnc/bug315-2-fix : Commit f27301c8230 in master
Branch linuxcnc/bug315-fix : Commit 13273aadcc9 in master
Branch linuxcnc/c-morley-patch-1 : Commit 96e52ac9af9 in master
Branch linuxcnc/compiler-fixes : Commit 6ba46eba262 in master
Branch linuxcnc/cradek/2.6/bug430-fix : Commit 9eeb2666434 in master
Branch linuxcnc/debian/RemovePatches : Commit 285463d9390 in master
Branch linuxcnc/fix-627 : Commit b522453237b in master
Branch linuxcnc/glo/zultron/remap-io-lcnc-2.6 : Commit e007416577a in master
Branch linuxcnc/gmoccapy_1_0 : Commit 5ccf94a91e1 in master
Branch linuxcnc/gmoccapy_JA_based_on_master : Commit 02086975991 in master
Branch linuxcnc/gmoccapy_additional_keyboard_shortcuts : Commit 03fddd2c161 in master
Branch linuxcnc/gtk3-halscope : Commit 336ab2d4f39 in master
Branch linuxcnc/hm2-features-for-2.6 : Commit 85d30465f10 in master
Branch linuxcnc/hotfix/g61-exactstop : Commit 663119abb3a in master
fatal: No tags can describe '3f97a52cb12ec7150554f5db1dfed28bf45816bd'.
Try --always, or create some tags.
Branch linuxcnc/ioctl_branch : Commit  in master
Branch linuxcnc/jepler/2.7/docbook-xsl : Commit f7bda079c0f in master
Branch linuxcnc/jepler/2.7/gmoccapy-i18n : Commit 8c37585293b in master
Branch linuxcnc/jepler/2.7/image2gcode-numpy : Commit 3d2aec7b18a in master
Branch linuxcnc/jepler/2.7/readline-license-check : Commit cbe65172bcd in master
Branch linuxcnc/jepler/2.7/remove-heuristic-delay-warning : Commit ec56cef05b4 in master
Branch linuxcnc/jepler/2.7/stretch : Commit c98aa619706 in master
Branch linuxcnc/jepler/2.8-cxxflags : Commit 9ff9036643e in master
Branch linuxcnc/jepler/bump-gcc-std : Commit 456b6d4609d in master
Branch linuxcnc/jepler/debian-configure-with-python27 : Commit 0310559763a in master
Branch linuxcnc/jepler/for-2.7 : Commit e24a899435a in master
Branch linuxcnc/jepler/hal-streams : Commit 3df5e11c76a in master
Branch linuxcnc/jepler/hm2-aliases : Commit a6bd7d30147 in master
Branch linuxcnc/jepler/hm2-public-apis : Commit 326efa3f8cf in master
Branch linuxcnc/jepler/lessboost : Commit acb4fc71cf2 in master
Branch linuxcnc/jepler/master/configure-libdl : Commit e7804e88dcc in master
Branch linuxcnc/jepler/master/halcmd-cplusplus : Commit 9b390cdcab6 in master
Branch linuxcnc/jepler/master/ickalb-tp-personalities : Commit 80473377e5b in master
Branch linuxcnc/jepler/master/missing_clock_nanosleep : Commit 4f778f15798 in master
Branch linuxcnc/jepler/master/nicokid-stepconf : Commit 0b3ec8cfd29 in master
Branch linuxcnc/jepler/master/nikokid-stepconf : Commit 0b3ec8cfd29 in master
Branch linuxcnc/jepler/master/nml-tcp : Commit aebad414dd2 in master
Branch linuxcnc/jepler/master/stdint : Commit a67d00c4ea7 in master
Branch linuxcnc/jepler/modernize-autoconf-boost-python : Commit ac0cf10c98f in master
Branch linuxcnc/jepler/opengl-without-numpy : Commit 5da0ee5b393 in master
Branch linuxcnc/jepler/prefer-png-images-2.8 : Commit b5312529722 in master
Branch linuxcnc/jepler/python3-halmodule-32bit-bug : Commit 667b4ead6b6 in master
[... 13 lines cut ...]
Branch linuxcnc/pncconf-gtk3 : Commit 4fe346f7f25 in master
Branch linuxcnc/pncconf_fix_missing_ssport : Commit 4b43058d802 in master
Branch linuxcnc/po4a-english-adjustments : Commit 47c6e0dfd41 in master
Branch linuxcnc/pr135 : Commit 44377d84a08 in master
Branch linuxcnc/pr850 : Commit a2a6aa40903 in master
Branch linuxcnc/pycleanup : Commit 4435aac5ebb in master
Branch linuxcnc/random_toolchange : Commit e743c98f902 in master
Branch linuxcnc/rellenberg-618-catch2-unit-tests : Commit 666720451b8 in master
Branch linuxcnc/remove-freqgen : Commit 427cf6522b3 in master
Branch linuxcnc/rene-cleanup : Commit 03856ece427 in master
Branch linuxcnc/s_code_fix : Commit 79c0966259a in master
Branch linuxcnc/separate-g92 : Commit 82e63fed1ed in master
Branch linuxcnc/sim-spindle : Commit f74a48f9335 in master
Branch linuxcnc/spindle_ini_fix : Commit 9f146585db1 in master
Branch linuxcnc/ss-wrapped-rotary : Commit 4d07a4b8f81 in master
Branch linuxcnc/taskmodule : Commit 5ad2b7c30b6 in master
Branch linuxcnc/test-dh-install : Commit 3551cc3c11d in master
Branch linuxcnc/tlo_all_axes : Commit ccf5d7874a1 in master
Branch linuxcnc/touchy-macro : Commit 2010001fbeb in master
Branch linuxcnc/unsupport-old-distros-in-master : Commit 323a1e5adc2 in master
Branch linuxcnc/use-tkpng : Commit d21d9fdb95f in master
Branch linuxcnc/v2.4_branch : Commit 4a5dda1cc8c in master
Branch linuxcnc/zultron/glo-2.6-remap-startline-fix : Commit 13538958723 in master
Branch linuxcnc/zultron/spindle_overhaul : Commit 103122ac24f in master
```

andypugh, 2024-02-22:
Good Gitting! That seems like a good list to be starting with.

satiowadahc, 2024-02-22:
If you actually want to see what branches contain the commit

`itsthere = git branch --contains $commit`

`echo itsthere`