LinuxCNC/linuxcnc #2910: Linear A and B axis jog according to Angular velocity when defined in display section and using GMOCCAPY

SpamsterFly, 2024-02-27:
# The issue tracker is not a support forum
## Here are the steps I follow to reproduce the issue:

 1. Setup a GMOCCAPY configuration with Linear XYZAB axes and joints
 2. Define the following in the display section
[DISPLAY]
DEFAULT_ANGULAR_VELOCITY = 5.60
MIN_ANGULAR_VELOCITY = 0
MAX_ANGULAR_VELOCITY = 56.00

All Axis and joints are defined as linear
[AXIS_A]
TYPE = LINEAR

[JOINT_4]
TYPE = LINEAR

 4. When jogging A or B axis in GMOCCAPY, the moves are very slow. Increasing angular velocity affects the linear A and B axis.
 Changing UI to Axis and the A and B Axis move according to linear velocity. Removing Angular velocity from Display section results in GMOCCAPY results in A and B axis jogging according to linear velocity.

## This is what I expected to happen:
A and B axis should jog with respect to linear velocity unless they are defined as Angular axis/joint

## This is what happened instead:
A and B linear axis were bound by angular velocity and limits when jogging

Here is a community thread describing this issue...
https://forum.linuxcnc.org/gmoccapy/51475-a-and-b-axis-jog-very-slow-when-angular-velocity-is-specified-in-display-section

## It worked properly before this:

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):Debian 10 Buster
 * I am using this kernel version (shown by `uname -a`): 4.19.0-24-rt.amd64
 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):2.8.4-1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): GMOCCAPY
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): PP SD-PEX10005

andypugh, 2024-02-27:
Whilst I agree that this is unexpected, if you have extra linear axes, these would normally be U, V or W. 

What is the machine? 5 linear axes seems unusual?

SpamsterFly, 2024-02-27:
Knee mill with quill and rotary. Knee is A rotary is B, I was attempting to setup rotary and knee slowed down when I added angular velocity to Display.

SpamsterFly, 2024-02-27:
I changed my coordinates to XYZUA and it appears to not exhibit the problem. Still seems defective.
Thanks Andy

gmoccapy, 2026-10-28:
I think this is a very special case.
IMHO this issue should be closed.

Norbert

grandixximo, 2026-10-28:
It is fixed in master by #4581 and #4594

closed 2026-10-28 by grandixximo