LinuxCNC/linuxcnc #2939: G71 introduces a circle when using the D parameter (remaining distance to stock)

sundtek, 2024-03-15:
# The issue tracker is not a support forum

G17 G54 G40 G80 G90
G21
G54
G97
G61
g18 g64 g8
G1 F140
S700
M3
G4 P1
G0 Z14.4
#<_startX>=14.3
#<_startZ>=12.5
o100 sub
	G1 X#<_startX> Z#<_startZ>
        G1 X0.01
        G1 Z0 X14.325
        G1 X15
o100 endsub
G71 Q100 X#<_startX> Z#<_startZ> D0.25 I0.2 R1 
M2

that half circle is not correct.

[image]

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

andypugh, 2024-04-26:
Is D meant to be a "stock to leave" number? The docs for G71 are terrible, which annoys me as I wrote much better ones for a different implementation of the cycles. 

Does it always fail? Or is is sometimes usable?

sundtek, 2024-04-26:
I only tested it once and put the result here for the record until someone (or I) have more time to fix it.
D should be stock to leave yes. 
Once the stock to leave is removed it's okay -  the entire circle should not be there when stock to leave is used.

andypugh, 2024-07-13:
@sundtek Are you able to check if #3020 fixes the issue?

sundtek, 2024-07-14:
Hi,

I won't be able to test before the end of next month, I was running buster before but the upgrade to bookworm was problematic and introduced some issues with the nvidia graphiccard (those issues are fixed), but LinuxCNC (latest git) refuses to work it wouldn't load motmod.

rene-dev, 2024-07-14:
Operation not permitted sounds like a permission issue.
also, lxrt? which kernel are you using?

sundtek, 2024-07-14:
I don't know anymore I just upgraded it in a hurry but did not have time to investigate the last issues... sorry I'm not at the location where this machine is until the end of August (and it's not online either).

andypugh, 2024-07-20:
I merged the fix into master last night, it should be there now.

sundtek, 2024-07-21:
ok I will test it tonight.

andypugh, 2024-08-13:
Fixed?

closed 2024-08-13 by andypugh

sundtek, 2024-10-02:
Sorry it took some time I can confirm that the issue is fixed.