LinuxCNC/linuxcnc #2948: parameter R in cycle g81-g82 ecc, in lathe not respect diametral position, only radial position  in g19 plane

powerio25, 2024-03-25:
use linuxcnc 2.9-pre-0 on uspace.
System Debian 10
-on lathe configuration with x-y-z and 2 spindle , when use X axis for life tool spindle use g19 plane.
the machine have limit on x plus 600mm and x minus -20 mm
The plus position value are up spindle centre , minus position value are down at spindle centre.
Normally we use only diametral mode G7
if stay for example in position x 500 mm in g7 diametral mode and command 
m3 s100$1
g19g94g81x400r420 f1000
the linuxcnc return message , max limit plus exceded, max max limit reaaly is 600mm
if command  and stay in position x500mm
m3s100$1
g19g94g81x400 r210 f1000  the machine work ok, executer rapit at 420 mm and after executed f1000 at x400 and retur rapid at 420mm.
the r parameter in g19 plane not respect diameter mode g7 , but use position in radial mode

This is valid on all macro cycle in plane g19 .
I have test is on g17 and run good.

andypugh, 2024-03-27:
My first thought is that using diameter mode for a canned cycle seems like an odd thing to do. It meant to be the retract distance above the surface in the direction along the cutter axis. 
I don't know if most people would expect the retract distance to consider radius/diameter mode in this case. 

Would you also expect to consider radius/diameter in the Y axis if using the G18 plane? 

I am more inclined to think that it is the X axis calculation that is wrong. 

I certainly agree that X and R should match in their behaviour, though.

powerio25, 2024-03-27:
if you work on a lathe, and work diametrically, the position commands e.g. g1 x60, they will be diametrical, the cnc will move the x axis to a diametrical position of 60, therefore in a macro cycle, the r must respect the same diametrical behavior as the position command.

At the moment when you execute g19 g81 x400 r 210 f100 , the r has a value half of that where the cycle will start and that is 420 .
If in the current state I wrote g19 g81 x400 r420 f100, lxcnc returns
that R is beyond the positive software limit, because it interprets it as R840 and since my limit is 600 mm positive, the problem occurs.

I raise this bug, because on other cncs, the R is concordant with the x in g19, as it is with the z in g17 and with the y in g18

The cycle g17 g81 z400 r420 f100 works correctly, in this plane, the R is greater than Z.

I will test the g18 plan and update you.

I think that the behavior of the g18 and g19 planes if I turn, the R word must be conditioned to the radial or diametric mode.

powerio25, 2024-03-27:
HI.
I have test on g18 plane : g18 g94 y-40 r-20 f100 and run good .
Rapid at -20 and f100 at -40 , return at -20.

The problem is only g19 plabe when use x axis and r word.