LinuxCNC/linuxcnc #2951: Message: "Ignoring task mode change while jogging"

tpa58, 2024-03-27:
I runing simulate example sim.gscreen or sim.gscreen.silverdragon.
After machine ON, homing axis, select axis and jogging mode, I try press GUI button + or  -.
Axis moving, but every click i get error message "Ignoring task mode change while jogging".
LinuxCNC 2.9.2 on VirtualBox and real PC.
[image]

Artem-Viveritsa, 2024-08-21:
I confirm, the same problem. It happens exactly when the key is released and gscreen calls
`self.gscreen.do_key_jog(_X, direction, False)`

Thus it calls jog while the other is still in progress. If the movement is incremental and the key is not released until the movement is complete, then there is no error. In the source code it looks like this:
```
int emcTaskSetMode(EMC_TASK_MODE mode)
{
    int retval = 0;

    if (jogging_is_active()) {
        emcOperatorError("Ignoring task mode change while jogging");
        return 0;
    }
...
```

I thought the problem was with my custom panel implementation, but the standard one has the same problem. Any ideas on how to fix this?

Artem-Viveritsa, 2024-08-21:
It's all about this line: `self.emc.jogging(1)`

I commented it out and the errors stopped appearing

```
    def do_key_jog(self,axis,direction,action):
        if self.data._JOG in self.check_mode(): # jog mode active:
                    if not action: cmd = 0
                    elif direction: cmd = 1
                    else: cmd = -1

                    self.emc.jogging(1) <-- HERE

                    #print(self.data.jog_increments[self.data.current_jogincr_index])
                    if self.data.jog_increments[self.data.current_jogincr_index] == ("continuous"): # continuous jog
                        #print("active axis jog:",axis)
                        self.emc.continuous_jog(axis,cmd)
                    else:
                        #print("jog incremental")
                        if cmd == 0: return # don't want release of button to stop jog
                        self.mdi_control.mdi.emcstat.poll()
                        if self.mdi_control.mdi.emcstat.state != 1: return
                        jogincr = self.data.jog_increments[self.data.current_jogincr_index]
                        distance = self.parse_increment(jogincr)
                        self.emc.incremental_jog(axis,cmd,distance)
```

Is it necessary if we are already in JOG mode?

andypugh, 2024-08-29:
Sorry @c-morley, but I think that you are best placed to decide on this one?

c-morley, 2024-08-31:
I guess one question to ask is:
is the message "Ignoring task mode change while jogging" useful?
If we remove the message the request for manual mode would be silently ignored as I think it should be.
Opinions?

Artem-Viveritsa, 2024-08-31:
> Opinions?

If the problem is in gscreen and we know what causes the error, then adjusting emc to it by simply suppressing errors would be wrong. I think the best option would be to fix the JOG operation in gscreen so that it more closely matches the logic of emc.

c-morley, 2024-08-31:
Well the problem could be both - a useless annoying error message and setting manual when it's already in manual.
I'm interested in fixing both unless there is a good reason to let the user know of an ignored task switch.
It seems to me at the most it should be a debug message, not an operator message.

Artem-Viveritsa, 2024-08-31:
Yes, lowering the status of this message to "debug" sounds reasonable. But so that it can be seen when setting DEBUG = 1 in the ini file, right?

As for gscreen, on key release events we can specify the handler and not call
`self.emc.continuous_jog(axis, False)` or, perhaps, `self.emc.stop_jog(axis) `

c-morley, 2024-08-31:
I would consider removing the message completely. I can't see how it's useful.

Key releases for continuous jog need to stop the axis, key release from interval jogging must not stop the axis.

While yes we can rearrange the code to only call emc.jogging() on press events, it's probably easier to just have emc.jogging() check for manual mode before changing it. I would bet it's possible to get the error message on press events if the machine is already in manual mode. Or we can do both things. I would prefer to do the least amount of change in released code.
Especially since I haven't been in this code for a very long time.

Artem-Viveritsa, 2024-09-02:
I settled on this option for myself:

Since this check ensures that in addition to JOG mode we are also in MAN mode
```
    def check_mode(self):
        """This function checks if Gscreen is in jog mode and manual mode
            Tries to call check_mode() in a handler file instead of this default function
            Requires a notebook widget called notebook_main
            Assumes page 0 is the manual page
            Requires a jog mode toggle button called button_jog_mode
            it returns a string list
        """
        try:
            return self.handler_instance.check_mode()
        except:
            pass
        string=[]
        if self.data.mode_order[0] == self.data._MAN and self.widgets.notebook_main.get_current_page() == 0:
            string.append( self.data._MAN)
            if self.widgets.button_jog_mode.get_active(): # jog mode active
                string.append(self.data._JOG)
        return string
```

The following method I got rid of unnecessary mode settings
```
    def do_key_jog(self,axis,direction,action):
        if self.data._JOG in self.check_mode(): # MAN and JOG mode active:
            if not action: cmd = 0
            elif direction: cmd = 1
            else: cmd = -1

            if self.data.jog_increments[self.data.current_jogincr_index] == ("continuous"):
                if cmd == 0:
                    self.emc.stop_jog(axis)
                    return
                self.emc.continuous_jog(axis,cmd)

            else:
                if cmd == 0: return # don't want release of button to stop jog

                jogincr = self.data.jog_increments[self.data.current_jogincr_index]
                distance = self.parse_increment(jogincr)

                self.emc.incremental_jog(axis,cmd,distance)
```

What do you think?

andypugh, 2024-09-02:
> I would consider removing the message completely. I can't see how it's useful.

I think I agree. Messages to the user are only useful if they relate to something that the user has done. And the user will be unaware that they even caused a task mode change, or even what that is.

c-morley, 2024-10-09:
I put this in master and we will see if anyone complains:
https://github.com/LinuxCNC/linuxcnc/commit/52dc209ff2888b62ea14a1f6b290d447fe8331b1