LinuxCNC/linuxcnc #2956: EMC_MDIhistory does not respect the requirement of the NIST standard

zz912, 2024-04-01:
Branche 2.9 RIP instalation
Tested in Gmoccapy (This issue may also affect other GUIs.)

Here is citation from NIST:
> (rs274 v3 - 3.6.1)
> Note: It is OK to program M0 and M1 in MDI mode, but the effect will probably not be noticeable, because normal behavior in MDI mode is to stop after each line of input anyway.

[Source](https://forum.linuxcnc.org/gmoccapy/51209-we-allow-stacking-of-mdi-commands-is-it-ok?start=10#289694)

HAL component EMC_MDIhistory [allow](https://forum.linuxcnc.org/gmoccapy/51209-we-allow-stacking-of-mdi-commands-is-it-ok?start=0 ) 'stacking' of MDI commands. So normal behavior in MDI mode is NOT to stop after each line of input anyway.
[image]
The second problem is that some standards require MDI commands to be executed only with a physical button.
EMC_MDIhistory can execute commands 3 ways:
1) Click on arrow
2) Write command and press enter
3) Double click on previous command

I propose a solution:
1) Add HAL pin to GUI to execute MDI command
The LinuxuCNC integrator will produce a physical button.
The signal from the physical button and the HAL pin halui.program.is-idle will be connected to the HAL component and2. This will ensure that the physical button is ignored during MDI command execution.
2) Prohibition of executing MDI commands by click on arrow, write command and press enter, double click on previous command.
This disable MUST be configurable.
In Gmoccapy it might look like this:
[image]

Sigma1912, 2024-04-01:
Are you filing this issue simply because the behavior is not as described in the NIST document or does this create other issues?

zz912, 2024-04-01:
> Are you filing this issue simply because the behavior is not as described in the NIST document 
Yes

> or does this create other issues?
I can't build a machine that meets the standards.

It is not standard software bug.

Sigma1912, 2024-04-02:
So do you see this as a GUI issue or an issue with the underlying code base?

I came across another violation of the NIST standard recently regarding G52/G92 deactivation on M2/M30:
https://github.com/LinuxCNC/linuxcnc/pull/2921

Personally I find it rather unfortunate that a standard configuration of LinuxCNC does not follow the standard NIST behavior.  
The problem is that these violations are not necessarily bugs but can be intentional. In the case of the nonstandard G52/G92 behavior it is actually tested for when we run the build tests. In the case of MDI stacking it is likely the same.

zz912, 2024-04-02:
> So do you see this as a GUI issue or an issue with the underlying code base?

Can you explain me what is "underlying code base"?

> Personally I find it rather unfortunate that a standard configuration of LinuxCNC does not follow the standard NIST behavior.
> The problem is that these violations are not necessarily bugs but can be [...]

Some people wants stacking MDI stacking some people like me wants the same behavior like NIST. 
https://forum.linuxcnc.org/gmoccapy/51209-we-allow-stacking-of-mdi-commands-is-it-ok?start=0
It must be configurable.

Sigma1912, 2024-04-03:
What I meat is, should the individual GUIs be responsible whether stacking is allowed or not or should the python interface do it?

zz912, 2024-04-03:
The LCNC integrator should be responsible for stacking MDI commands or not stacking. A person who makes a machine.

Small printer with small stepper motor - it doesnt matter.
Big machine with servos - I recommeneded use standards.

zz912, 2024-04-08:
Hello,

I don't want to debate whether the MDI queue is a matter of the NIST standard or another standard. Let's leave that to the LCNC integrators.

I would like to recap the possibilities.

**Simple (no graphic change):**
1) disable the MDI queue
2) enable MDI queue
3) make disable/enable MDI queue configurable (I prefer this)

**More complex (graphical change):**

4) enter blocks of MDI commands (This is the best solution)

This is how a friend has it on an old CNC machine with a sinumerik from the 80s. It still has a green CRC screen.
He can enter several lines of MDI commands into the MDI screen.
Once he enters this, he starts this sequence with the physical green button. He would like this solution in the current LCNC as well.
In Gmoccapy's implementation, it might look like this:
[image]
Currently, the Enter key executes the current line of MDI code.
If we decided on this solution, the Enter key should create another line. The execution of this block would be the blue arrow, or just the HAL pin. MDI blocks queuing is disabled.

zz912, 2024-04-09:
I'm not an expert on NIST standards. That's why I prefer configurability. On the other hand, I'm not the only one who prefers to ban the MDI queue.

terrylm, 2025-04-05:
Hi, I happen to like the old MDI queue. I used it a lot, often 3 or four blocks ahead of the one running, end with an M5 so that when the spindle stops I know to go back and check it. Please make it configurable! While I understand that anyone building machines for sale will want to follow standards so they can not be sued. For myself I am not concerned with strict adherence to a 25 year old or more standard, or any standard. I am more interested it what works, and a queue works.

So please make it configurable, please.