LinuxCNC/linuxcnc #2966: Glade Tooltable widget destroys tooltable in a particular case

hansu, 2024-04-20:
When using locale settings for e.g. Germany, so having a "," as decimal point, the tool table shows the correct decimal point after starting.
But after loading an (empty) G-code file and reloading the tool table, the decimal point changes to a dot:

https://github.com/LinuxCNC/linuxcnc/assets/67957916/421d6f9a-7a8d-4f6c-90cb-40ac777b398f

This can lead to loss of the complete tool table in the following (quite common) case:

1. Load a G-code file
2. Add the missing tool
3. Apply the changes ... boom! all tools are gone!

https://github.com/LinuxCNC/linuxcnc/assets/67957916/a59e732b-f6b2-4aa1-a6b7-9ae81574fa16

Version: latest from 2.9 branch.

```
Traceback (most recent call last):
  File "/home/cnc/linuxcnc/linuxcnc-2.9/bin/gmoccapy", line 4947, in on_btn_apply_tool_changes_clicked
    self.widgets.tooledit1.save(None)
  File "/home/cnc/linuxcnc/linuxcnc-2.9/lib/python/gladevcp/tooledit_widget.py", line 348, in save
    line = line + "%s%s "%(KEYWORDS[num], locale.atof(test))
                                          ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/locale.py", line 338, in atof
    return func(delocalize(string))
           ^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: '0,0000'
 (gmoccapy:62)
Tool file was modified since it was last read
/usr/lib/python3/dist-packages/gi/overrides/GObject.py:493: Warning: ../../../gobject/gsignal.c:2772: instance '0x460ff90' has no handler with id '113342181'
  return func(*args, **kwargs)
```

zz912, 2024-04-23:
I can confirm this behavior. Any G-code file can be used for recall, not just an empty G-code file.

In place for tool information is used as a decimal point and a dot. It is also in the Czech localization.
[image]
It looks wierd.

hansu, 2024-06-28:
One possible workarounds are:

hansu, 2024-06-28:
The problem does not appear on Debian Bullseye / Python 3.9.2

hansu, 2024-06-28:
I was able to encircle the error. The problem is that the locale setting gets destroyed when showing the main box.

Good to see with these debug prints:
https://github.com/hansu/linuxcnc/commit/0fe0482b10a29801de4deacfb1f438e90dd1139e

Output:
```
--- INIT:352, locale:  ('de_DE', 'UTF-8')
--- INIT:356, locale:  (None, None)
```
If I set the locale again after this with
`locale.setlocale(locale.LC_ALL, '')`
all works fine.
But this does't sound like a clean solution.

@c-morley any idea on this?

hansu, 2024-07-06:
Just for information to avoid double work - my working branch: https://github.com/hansu/linuxcnc/tree/2966-gmoccapy-destroys-tooltable

zz912, 2024-07-06:
Hello,

I tested your branche. 
Backup file is working. It is fine.

BUT, I have different results from your test:
branche hansu/2966-gmoccapy-destroys-tooltable
--- INIT:352, locale:  ('cs_CZ', 'UTF-8')
--- INIT:356, locale:  ('cs_CZ', 'UTF-8')

branche linuxcnc/master (I added "print ...." commands to source code)
--- INIT:352, locale:  ('cs_CZ', 'UTF-8')
--- INIT:356, locale:  ('cs_CZ', 'UTF-8')

I use RIP installation.

zz912, 2024-07-06:
I tried change localization and result is same:
--- INIT:352, locale:  ('de_DE', 'UTF-8')
--- INIT:356, locale:  ('de_DE', 'UTF-8')

commands for set localization:
`localectl set-locale LANG=cs_CZ.UTF-8`

`localectl set-locale LANG=de_DE.UTF-8`

`localectl set-locale LANG=en_US.UTF-8`

`localectl`

source: https://www.server-world.info/en/note?os=Debian_12&p=locale

hansu, 2024-07-06:
Propably the locale setting get lost at another position on your system if you still encounter the tooltable bug.
You can set the debug print on other lines to figure out when it happens. Maybe begin with placing it at the end of the gmoccapy init function.

zz912, 2024-07-06:
I don't understand why you are looking for bugs in the init function.

I added the command "print ..." twice to the on_btn_new_clicked function:
```
gmoccapy.py:

    # make a new file
    def on_btn_new_clicked(self, widget, data=None):
        tempfilename = os.path.join(_TEMPDIR, "temp.ngc")
        content = self.get_ini_info.get_RS274_start_code()
        content += "\n\n\n\nM2"
        print("on_btn_new_clicked-1", locale.getlocale(locale.LC_NUMERIC))
        gcodefile = open(tempfilename, "w")
        gcodefile.write(content)
        gcodefile.close()
        if self.widgets.lbl_program.get_label() == tempfilename:
            self.widgets.hal_action_reload.emit("activate")
        else:
            self.widgets.hal_action_open.load_file(tempfilename)
            # self.command.program_open(tempfilename)
        self.widgets.gcode_view.grab_focus()
        self.widgets.btn_save.set_sensitive(False)
        print("on_btn_new_clicked-2", locale.getlocale(locale.LC_NUMERIC))
```

To execute this function, you must press this button:
[image]

Result after the first press:
on_btn_new_clicked-1 ('cs_CZ', 'UTF-8')
on_btn_new_clicked-2 ('cs_CZ', 'UTF-8')

Result after the second press:
on_btn_new_clicked-1 (None, None)
on_btn_new_clicked-2 (None, None)

Something delocalizate it, but I don't know what happens after loading some *.ngc file. Could it be a gremlin? I couldn't get Gmoccapy to run without Gremlin.

zz912, 2024-07-06:
If I comment this line:
[            self.widgets.hal_action_open.load_file(tempfilename)](https://github.com/LinuxCNC/linuxcnc/blob/20546d065c819a0b9dd71e7405f8c3a29a92ea22/src/emc/usr_intf/gmoccapy/gmoccapy.py#L5292)

Gmoccapy is not delocalizitied after on_btn_new_clicked

hansu, 2024-07-06:
>I don't understand why you are looking for bugs in the init function.

Because I wanted to figure out when ilthe locale gets lost. But does not seem to happen at a specific position.

hansu, 2024-07-11:
> If I comment this line: [ self.widgets.hal_action_open.load_file(tempfilename)](https://github.com/LinuxCNC/linuxcnc/blob/20546d065c819a0b9dd71e7405f8c3a29a92ea22/src/emc/usr_intf/gmoccapy/gmoccapy.py#L5292)
> 
> Gmoccapy is not delocalizitied after on_btn_new_clicked

This does not really resolves it.

zz912, 2024-07-12:
> > If I comment this line: [ self.widgets.hal_action_open.load_file(tempfilename)](https://github.com/LinuxCNC/linuxcnc/blob/20546d065c819a0b9dd71e7405f8c3a29a92ea22/src/emc/usr_intf/gmoccapy/gmoccapy.py#L5292)
> > Gmoccapy is not delocalizitied after on_btn_new_clicked
> 
> This does not really resolves it.

1) I agree that this is not a solution to the main problem. Please consider my posts as an attempt to help you.This line here in the code does something. I don't know exactly what. It would be stupid to just delete it.

2) I found out that the main problem occurs when the LCNC is in the state:
(None, None)
You solved it with this:
locale.setlocale(locale.LC_ALL, '') in tooledit_widget.py

3) Command self.widgets.hal_action_open.load_file causes something that delocalizes LCNC:
('cs_CZ', 'UTF-8') => (None, None)
Unfortunately, my skills are not good enough to get deeper into the problem.
If someone could explain to me what the self.widgets.hal_action_open.load_file command does and where its code is located, that would help.
Have you tried deleting this line and checking the localization on your computer?
Maybe we have a race-condition again and the problem is not in self.widgets.hal_action_open.load_file. I wonder how it behaves on another PC.

4) It's good that you fixed the main problem, but it would be good to find the cause. Writing "everywhere" locale.setlocale(locale.LC_ALL, '') is not a long-term solution.

hansu, 2024-07-12:
Ok I did some more investigation and the reason why the localisation gets lost for me during the initialisation is because I have chosen to load a file on startup in the preferences.

I did some more investigations and found out that `hal_action_open.load_file()` is not directly the problem.
But it happens in conjunction with that.
If you add debug prints just after calls to this functions, the locale is still valid.
It happens when you go back after creating a file or switch to another mode after loading a file, 
so in `on_ntb_button_switch_page()` (or somewhere in between).
So basically when switching the context. But only when a file is loaded.
I have no idea why `self.linuxcnc.program_open(filename)` causes that when switching tabs. 

>If someone could explain to me what the self.widgets.hal_action_open.load_file command does and where its code is located, that would help.

`lib/python/gladevcp/hal_filechooser.py:111`

zz912, 2024-07-12:
> It happens when you go back after creating a file or switch to another mode after loading a file, ...

I dont agree with you. It must be something that happens automatically.

I made modification of cooling button in  gmoccapy.py for better testing:

```
# =========================================================
# Coolant an mist coolant button
    def on_tbtn_flood_toggled(self, widget, data=None):
        print("on_tbtn_flood_toggled-1", locale.getlocale(locale.LC_NUMERIC))
        self.widgets.hal_action_open.load_file("../nc_files/3D_Chips.ngc")
        print("on_tbtn_flood_toggled-2", locale.getlocale(locale.LC_NUMERIC))
```

and deactivate function _update_coolant
```
    def _update_coolant(self):
        a=5
```

Result after first click:
```
on_tbtn_flood_toggled-1 ('cs_CZ', 'UTF-8')
on_tbtn_flood_toggled-2 ('cs_CZ', 'UTF-8')
```

Result after second click:
```
on_tbtn_flood_toggled-1 (None, None)
on_tbtn_flood_toggled-2 (None, None)
```

hansu, 2024-07-12:
Yes when some context changes or some containers are being reloaded does this happen. Initiated manually or not does not really seem to make a difference here.

zz912, 2024-07-14:
I found source of problem:

If I deactivate function load in gremlin.py
https://github.com/LinuxCNC/linuxcnc/blob/d1414e63eb19f2ebb6c300ea40b4404869c516a3/src/emc/usr_intf/gremlin/gremlin.py#L310-L342

=>

```
    def load(self,filename = None):
        print("gremlin_load-1")
```

LCNC is not delocalizated

hansu, 2024-07-14:
Not sure if this is the source of the problem. Have you tried Gscreen? That GUI uses the same tooltable widget and seem to not have this problem...

zz912, 2024-07-15:
Gscreen has the same problem. Gremlin.py has no "import locale".

hansu, 2024-07-15:
Did you manage to crash the tool table in Gscreen in the same way?

zz912, 2024-07-16:
No, I am interesting only about delocation now:
('cs_CZ', 'UTF-8')
=>
(None, None)

zz912, 2024-07-20:
I added to gremlin.py:
```
# localization
import locale
BASE = os.path.abspath(os.path.join(os.path.dirname(sys.argv[0]), ".."))
LOCALEDIR = os.path.join(BASE, "share", "locale")
locale.setlocale(locale.LC_ALL, '')
```

I added the "print" commands to the "load" function:
```
    def load(self,filename = None):
        print("gremlin_load-1", locale.getlocale(locale.LC_NUMERIC))
        s = self.stat
        s.poll()
        if not filename and s.file:
            filename = s.file
        elif not filename and not s.file:
            return

        td = tempfile.mkdtemp()
        self._current_file = filename
        print("gremlin_load-2", locale.getlocale(locale.LC_NUMERIC))
        try:
            random = int(self.inifile.find("EMCIO", "RANDOM_TOOLCHANGER") or 0)
            canon = StatCanon(self.colors, self.get_geometry(),self.lathe_option, s, random)
            parameter = self.inifile.find("RS274NGC", "PARAMETER_FILE")
            temp_parameter = os.path.join(td, os.path.basename(parameter or "linuxcnc.var"))
            if parameter:
                shutil.copy(parameter, temp_parameter)
            canon.parameter_file = temp_parameter
            print("gremlin_load-3", locale.getlocale(locale.LC_NUMERIC))
            unitcode = "G%d" % (20 + (s.linear_units == 1))
            print("gremlin_load-4", locale.getlocale(locale.LC_NUMERIC))
            initcode = self.inifile.find("RS274NGC", "RS274NGC_STARTUP_CODE") or ""
            print("gremlin_load-5", locale.getlocale(locale.LC_NUMERIC))
            result, seq = self.load_preview(filename, canon, unitcode, initcode)
            print("gremlin_load-6", locale.getlocale(locale.LC_NUMERIC))
            if result > gcode.MIN_ERROR:
                self.report_gcode_error(result, seq, filename)
            self.calculate_gcode_properties(canon)
            print("gremlin_load-7", locale.getlocale(locale.LC_NUMERIC))
        except Exception as e:
            print (e)
            self.gcode_properties = None
        finally:
            shutil.rmtree(td)

        self.set_current_view()
```

Result:
```
gremlin_load-1 ('cs_CZ', 'UTF-8')
gremlin_load-2 ('cs_CZ', 'UTF-8')
gremlin_load-3 ('cs_CZ', 'UTF-8')
gremlin_load-4 ('cs_CZ', 'UTF-8')
gremlin_load-5 ('cs_CZ', 'UTF-8')
gremlin_load-6 (None, None)
gremlin_load-7 (None, None)
```

This implies that the problem is here:
https://github.com/LinuxCNC/linuxcnc/blob/732919b3974666b45644f54407076fbee0e5459a/src/emc/usr_intf/gremlin/gremlin.py#L331

Here is the edited file:
[gremlin.py.zip](https://github.com/user-attachments/files/16319217/gremlin.py.zip)

hansu, 2024-07-20:
> ```
> gremlin_load-6 (None, None)
> gremlin_load-7 (None, None)
> ```

You still get this when you add the import locale ... to gremlin?

zz912, 2024-07-20:
It continue to glcanon.py

I added:
```
import sys

# localization
import locale
BASE = os.path.abspath(os.path.join(os.path.dirname(sys.argv[0]), ".."))
LOCALEDIR = os.path.join(BASE, "share", "locale")
locale.setlocale(locale.LC_ALL, '')
```

Modify:
```
    def load_preview(self, f, canon, *args):
        print("glcanon_load_preview-1", locale.getlocale(locale.LC_NUMERIC))
        self.set_canon(canon)
        print("glcanon_load_preview-2", locale.getlocale(locale.LC_NUMERIC))
        result, seq = gcode.parse(f, canon, *args)
        print("glcanon_load_preview-3", locale.getlocale(locale.LC_NUMERIC))
        if result <= gcode.MIN_ERROR:
            self.canon.progress.nextphase(1)
            canon.calc_extents()
            self.stale_dlist('program_rapids')
            self.stale_dlist('program_norapids')
            self.stale_dlist('select_rapids')
            self.stale_dlist('select_norapids')
        print("glcanon_load_preview-4", locale.getlocale(locale.LC_NUMERIC))
        return result, seq
```

Result:
```
glcanon_load_preview-1 ('cs_CZ', 'UTF-8')
glcanon_load_preview-2 ('cs_CZ', 'UTF-8')
glcanon_load_preview-3 (None, None)
glcanon_load_preview-4 (None, None)
```

This implies that the problem is here:
https://github.com/LinuxCNC/linuxcnc/blob/c2570af9a44907e151eb5b54770fc1d76b07f06f/lib/python/rs274/glcanon.py#L2004

Here is the edited file:
[glcanon.py.zip](https://github.com/user-attachments/files/16319389/glcanon.py.zip)

> > ```
> > gremlin_load-6 (None, None)
> > gremlin_load-7 (None, None)
> > ```
> 
> You still get this when you add the import locale ... to gremlin?

Yes

zz912, 2024-07-20:
@fragmuffin Do you think it's possible for gcode to cause delocalization?

zz912, 2024-07-20:
In glcanon.py is 
`import gcode`

Where is source?
I found only linuxcnc-2.9/lib/python/gcode.so

zz912, 2024-07-21:
Hi Hans,

I assume the file linuxcnc-2.9/lib/python/gcode.so comes from source code written in 'C' language.

I did research about "locale + Python language + C language". I found these two resources:
Here's a guy who has a similar problem and also didn't find any nice solution: 
https://stackoverflow.com/questions/140295/using-locale-setlocale-in-embedded-python-without-breaking-file-parsing-in-c -thr
Here is the locale documentation: 
https://docs.python.org/3/library/locale.html

My conclusions:

1) setlocale()  is not thread-safe

In documentation:

> [setlocale()](https://docs.python.org/3/library/locale.html#locale.setlocale) is not thread-safe on most systems. Applications typically start with a call of
> 
> import locale
> locale.setlocale(locale.LC_ALL, '')
> 
> This sets the locale for all categories to the user’s default setting (typically specified in the LANG environment variable). If the locale is not changed thereafter, using multithreading should not cause problems.

LinuxCNC is a multithreading project, so using the locale library is inappropriate for LinuxCNC. On the other hand, there is probably no better solution.
Therefore, I suppose we will forever be fighting localization and delocalization.

2) I assume that without drastic changes, it will not be possible to simply switch localizations. Therefore, a simple rule will apply:
locale settings LCNC = default locale settings OS

----------------------------------------------------------------------------------------------------
If my assumption that we will always be at war with localization and delocalization is correct, then we should:
- point to it in the source codes.
- remove the delocalization problem at the lowest level. At the same time, assume that delocalization can also occur at a higher level of code.

In this case, I fixed the problem as low as possible in this PR:
https://github.com/LinuxCNC/linuxcnc/pull/3036

On the other hand, your PR should also be merged:
https://github.com/LinuxCNC/linuxcnc/pull/3026
Because we don't know if some other delocalization problem will arise in the future.

I have some comments on your commit

https://github.com/LinuxCNC/linuxcnc/pull/3026/commits/c5a4ea4b8346a9a8c5d69d7809286624bd726466
-  No one knows why locale.setlocale(locale.LC_ALL, '') is used here.
I would add a comment: "For the correct functionality of the following code, it is necessary to have a correctly set localization"
- Even better would be to check if:
locale.getdefaultlocale == locale.getlocale
Only after this check use locale.setlocale(locale.LC_ALL, '')
At the same time add Debug message: [glacanon.py] LCNC delocalization occurred and now localization occurred

hansu, 2024-07-28:
Fixed by https://github.com/LinuxCNC/linuxcnc/pull/3041

But what do you think about backing up the tool table before saving and the additional checks?

closed 2024-07-28 by hansu

zz912, 2024-07-28:
Can I ask you for https://github.com/LinuxCNC/linuxcnc/issues/2966#issuecomment-2073139614 ?

It is necessary to locale Offset Z +Vcc.

I don't know if this bug is related to the locale library.

hansu, 2024-07-28:
Do you mean the offset is still with a dot instead of a comma?

zz912, 2024-07-28:
Yes, the same problem has label Vcc.

hansu, 2024-07-28:
Ah ok you mean Vc - because Vcc is something completely different ;-)

zz912, 2024-07-28:
Sorry - Vc.

hansu, 2024-07-28:
If we have localized values in the tool info area, then maybe the increments should also be localized? But then the DRO wouldn't fit the format of the increments so I added localization there as well. But it looks really weird to have commas there. So I would tend to have only localized values for tool related values.

[image]

zz912, 2024-07-29:
I don't know if it looks weird. Anyway, it's new. Would you please ask a question on the discussion forum? Should we use comma or dot?

gmoccapy, 2024-07-29:
I do not like the DRO with comma as decimal separator. 
Have you looked at oteh machines? 
IMHO no machine producer changes the decimal separator for the DRO.

IMHO: All Info stuff should remain as it was, also Vc and Toolinfo and DRO
Introducing locale settings everywhere is not easy to maintain and does the comma bring additional information?

I only introduced the localization, to be able to use German keyboard.

Norbert

zz912, 2024-07-29:
Tool diameter has comma. Next Tool offset has dot.
What now Norbert?
Change Tool diameter to dot?

hansu, 2024-07-29:
> I do not like the DRO with comma as decimal separator. 

I totally agree  :)

> IMHO: All Info stuff should remain as it was, also Vc and Toolinfo and DRO Introducing locale settings everywhere is not easy to maintain and does the comma bring additional information?

I think the maintenance is not a problem -- it is more a matter of taste.
I only wanted to have it consistent because the diameter is localized here:

https://github.com/LinuxCNC/linuxcnc/issues/2966#issuecomment-2073139614
https://github.com/LinuxCNC/linuxcnc/blob/ae484723d9558cb78459468d839eb9497d5841a1/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3525

So I thought it was your intention to have localization here.

>I only introduced the localization, to be able to use German keyboard.

I also agree that localization is only necessary when entering a comma with the keyboard.
Furthermore I think the tool table should always use dots and convert the entered values based on the locale settings to a float. But I will not touch that..

gmoccapy, 2024-07-31:
Tool table:
I agree it should use dots. Isn't it? IMHO we do convert in both ways reading and displaying and also when values are entered. As I am not on my development PC I can not check that.

Norbert

hansu, 2024-07-31:
No it does not. See the videos in the issue description.

So as a first step I remove the localization for diameter?

zz912, 2024-08-14:
> So as a first step I remove the localization for diameter?

I agree.

zz912, 2024-08-17:
> https://github.com/LinuxCNC/linuxcnc/blob/ae484723d9558cb78459468d839eb9497d5841a1/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3525
> 
> So I thought it was your intention to have localization here.

I think it's different.
This line:
https://github.com/LinuxCNC/linuxcnc/blob/ae484723d9558cb78459468d839eb9497d5841a1/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3525
does not make a decimal point in the "tool diameter" number.

https://docs.python.org/3/library/locale.html :
> locale.atof(string, func=float)
>     Converts a string to a number, following the [LC_NUMERIC](https://docs.python.org/3/library/locale.html#locale.LC_NUMERIC) settings, by calling func on the result of calling [delocalize()](https://docs.python.org/3/library/locale.html#locale.delocalize) on string.

atof is only used to convert a string to a float so that it can be written to halcomp

This line must not be edited.

I think that if the Tooltable is converted to dots, the problem with the Tool diameter Label will be solved by itself.

hansu, 2024-08-19:
>  does not make a decimal point in the "tool diameter" number.

```

locale.atof(string, func=float)

    Converts a string to a number, following the [LC_NUMERIC](https://docs.python.org/3/library/locale.html#locale.LC_NUMERIC) settings, by calling func on the result of calling [delocalize()](https://docs.python.org/3/library/locale.html#locale.delocalize) on string.
```

\[ https://docs.python.org/3/library/locale.html ]

hansu, 2024-08-19:
Yes you are right, this label just picks a string from the tooltable widget.

We can have it consistent by adding this: https://github.com/hansu/linuxcnc/commit/1231fe7a315b2f95535be7b99d3456b37372a64b

zz912, 2024-08-19:
Or we can delocalize tooltable widget.

As far as I understand, the tooltable widget should only use locale to accept float numbers with a decimal comma. The Tooltable widget should display decimal points.

hansu, 2024-08-19:
>As far as I understand, the tooltable widget should only use locale to accept float numbers with a decimal comma. The Tooltable widget should display decimal points.

That's what Norbert said but I don't know Chris' intention who created this widget...

zz912, 2024-08-19:
What should we do now? We need Chris' intention. @c-morley

c-morley, 2024-08-19:
I'm from Canada... decimals are fine by me :)
I would have thought if one uses commas for decimals, one would expect them everywhere.

Then again gcode doesn't use commas as decimal.

I'm good with what you fellows feel is best.

gmoccapy, 2024-08-20:
Here in good old Germany entering values with comma as decimal separator is necessary, as the keyboard num pad will use that separator.
Showing the values with dots is not the problem, as long as it possible to edit the values in the cell itself.

Norbert

zz912, 2024-08-21:
So now we know what we want to do.
Norbert (author of Gmoccapy) approved it.
Chriss (author of Tooltable widget) approved it.
Question for Hans: Do you know how to do it?
Hans, I don't want to pressure you, it's about whether I should try. But I still can't do it myself, so it will be for a long time.

hansu, 2024-08-21:
Yeah I can do this but I wonder if it's worth the work. I am not really bothered by the commas in the tool table. 
There are other things in this widget I would like to.change more (e.g..allow only a single selection and/or use the selected line for the tool used for the tool change ...)

gmoccapy, 2024-08-21:
@hansu i do agree, that the whole tool stuff need a rework. I have been playing with Rene-Dev on a tool table based on sql and it did work. I know that Rene is redoing a lot on the tool part.

So IMHO we should only invest very little time to just fix the bug and wait for a reasonable tool handling version

Norbert

zz912, 2024-08-21:
> Yeah I can do this but I wonder if it's worth the work. I am not really bothered by the commas in the tool table.

What would be the motivation for you to do it? I am a poor and ugly boy.

Maintaining stable releases isn't fun or sexy, but ...

zz912, 2024-08-30:
Reading float with comma is Norbert's request.
https://github.com/LinuxCNC/linuxcnc/issues/2966#issuecomment-2298177492