LinuxCNC/linuxcnc #2967: Python oword freezes UI

TurBoss, 2024-04-21:
Hello

python OWORD freezes UI

## Here are the steps I follow to reproduce the issue:

This was tested on axis

 1. write a python `oword.py` file with this content
 
```python
import time

def test_sleep(self, n):
    print("Timer started")
    time.sleep(n)  # wait n seconds
    print("Timer ended")

```

 2. from a G-CODE file or MDI input run `o<test_sleep> call [10]`
 

## This is what I expected to happen:

 The UI to begin responsive while OWORD is waiting

## This is what happened instead:

The UI gets stuck until sleep ends  and is not possible to interact with any element

_____________________________________________

I`m not sure if this is an issue or its normal

Thanks

jethornton, 2024-04-22:
Python time.sleep() is a blocking call and all processing stops until it times out.

 ```
time.sleep(secs)

    Suspend execution of the calling thread for the given number of seconds. The argument may be a floating point number to indicate a more precise sleep time.

    If the sleep is interrupted by a signal and no exception is raised by the signal handler, the sleep is restarted with a recomputed timeout.

    The suspension time may be longer than requested by an arbitrary amount, because of the scheduling of other activity in the system.
```

c-morley, 2024-04-22:
But this python context is gcode(motion controller), the GUI should no be affected.as it's a different python Interpreter.
Something sounds not right.
Is there any other info you can give?

c-morley, 2024-04-22:
what GUI are you using?

jethornton, 2024-04-22:
Thanks Chris, I didn't know that the gcode was executed with a different python instance.

c-morley, 2024-04-22:
Well I may he miss understanding. I don't see in thr docs where an oword sub can be made in python.

jethornton, 2024-04-22:
Running Axis sim and creating the file and running the oword command I get:
EOF in file:usr/share/axis/images/axis.ngc
seeking o-word: o<test_sleep> from line: 0

andypugh, 2024-04-22:
Well, you probably _can_ run Python as an O-word sub, but it's not well documented. 

This works as a subroutine....

```
O<pytest> SUB
;py,import sys
;py,print(sys.version)
O<pytest> ENDSUB
M2
```

TurBoss, 2024-04-23:
hello, 

thanks for your responses

calling python from gcode sub also has the same issue

```ngc
o<pytest> sub

;py,import time
;py,time.sleep(10)

o<pytest> endsub

o<pytest> call
```

while this works

```ngc

o<sleep> sub

G1 x1 F10
G4 P10
G1 x2 F10

o<sleep> endsub

o<sleep> call
```

rmu75, 2024-04-26:
this seems to be a very obscure and unsafe feature.

```
o<pytest> sub

;py,while True: pass

o<pytest> endsub

o<pytest> call
```
I suspect the GUIs block at status update (ngc-interpreter is occupied in python and doesn't answer).