LinuxCNC/linuxcnc #2969: LinuxCNC 2.9.2 Pi 5  Debian Bookworm hdmi_force_hotplug not working

dqrw, 2024-04-28:
## Here are the steps I follow to reproduce the issue:

This looks to be a Bookworm issue but not sure how to submit that issue since this OS is specific to LinuxCNC.

Setup to run Pi 5 LinuxCNC in headless configuration.

 1. Set hdmi_force_hotplug=1 in boot/config.txt
 2. Install x11vnc and setup to startup with systemctl
 3. Setup to auto login in.
 4. Setup for wireless connectivity.
 5. Disable login timeout (never logout)
 6. Power down and remove all cables.
 7. Plug in power.
 8. Make sure you can ping from client.
 9. VNC into server with TigerVNC or equivalent.

## This is what I expected to happen:
I would expect to connect and be prompted for a password and to enter proper password. Then to be connected and see full display.

## This is what happened instead:

The client could connect and was not prompted for the password. The configuration works with a HDMI connection but is supposed to work without a display connected.

The setup also works properly without a HDMI monitor using the current Raspberry Pi OS.

## Information about my hardware and software:

LinuxCNC 2.9.2 Pi 5  Debian Bookworm 

Raspberry Pi 5 8 GB RAM. 256 GB NVme SSD.

Booted from both SSD and Micro SD card when testing.

andypugh, 2024-09-28:
Does this not happen with the stock Pi OS?

dqrw, 2024-09-28:
Hello,    Thank you for your response.
    I have just powered up my RPi 5 using the stock RPi 5 OS (Bookworm if I am not mistaken) that I updated last week and it properly functions without needing the dummy HDMI plugged in.
   Here is the current RPi 5 OS "uname -a" information below:

   Linux LinuxCNC 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux
   My main PC is using TigerVNC Viewer to IP into the headless RPi 5 system.
   Not sure if you require information for my main Host system but I have included that below "just in case":

   Linux LinuxCNC 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux

   Please let me know is you need me to perform any additional testing or additional information.
   Thank You and Best Regards
         David Q. R. Wagoner
Yahoo Mail: Search, Organize, Conquer

dqrw, 2024-09-28:
Correction!

I apologize but the "uname -a" information from my last email for my main Host system was pasted incorrectly. Below is the correct information:

Linux DQRWagoner-Ubuntu 6.8.0-45-generic #45-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug 30 12:02:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
    Best Regards

   David Q. R. Wagoner.

andypugh, 2024-09-30:
This sounds like there is something in the pi-specific kernel. It is likely to be a lot of work (it takes several hours to build a custom kernel on the Pi) to add a specific patch to the kernel that we put on the DS card image (at the moment we just pull down the stock kernel image from Debian). 
Do any of the patches here look to be the ones that add this capability? 
https://github.com/search?q=repo%3Araspberrypi%2Flinux+hdmi_force_hotplug&type=commits

Is this purely a VNC issue? I don't use VNC for remote Pi operation, I use ssh (ssh -Y pi@raspberrypi4 or similar command) to get a local X display, though this is not working as well as it used to now Wayland in in the game) 

At this point I see two issues:
1) How would we fix this? 
2) Is it worth it?

dqrw, 2024-10-01:
Hello,    Thank you for your reply.
    I will look at the link you provided and see if I can figure out how to get the test images to try. I'm not a Linux power user but I can usually fumble around and figure things out.
   As for "if the fix is worth it", I cannot pretend to know the LinuxCNC maintainers priorities but I will say that running headless with the tiny RPi5 footprint definitely has space savings advantages when not requiring a keyboard and display. Also, I did spend several frustrating days of reading and troubleshooting trying to get it working with the LinuxCNC kernal when it worked first time with the stock RPi5 Bookworm kernal. At the very least, an appnote documenting this issue might save someone else days of frustration and possibly giving up using this VERY useful device and configuration.
       Again, I will try to figure out how to test the patches you provided and let you know the results.
       Thank You and Best Regards        David Q R Wagoner        TinyCNC-II Project  
|  
|   
|   
|   |    |

   |

  |
|  
|   |  
TinyCNC-II Project
 
Build a Small 4-Axis CNC Controller for the Home Shop.
  |   |

  |

  |

  

Yahoo Mail: Search, Organize, Conquer 

https://github.com/search?q=repo%3Araspberrypi%2Flinux+hdmi_force_hotplug&type=commits

Is this purely a VNC issue? I don't use VNC for remote Pi operation, I use ssh (ssh -Y ***@***.*** or similar command) to get a local X display, though this is not working as well as it used to now Wayland in in the game)

   
   - How would we fix this?
   - Is it worth it?

andypugh, 2024-10-03:
Have you tried ssh instead? This doesn't need even a virtual monitor on the Pi. You basically launch the application from the remote machine, it runs on the Pi, and sends the graphical data back to your remote machine. I use it all the time. 

You need ssh set up on the Pi (but that's very useful anyway, as a way to do remote command-line things) 

The ssh -Y username@pi.local (or whatever) will give you a remote terminal prompt (executing on the Pi). 
And "linuxcnc" will start LinuxCNC, with all graphical windows appearing on your remote machine. 

You can also try ssh -X instead (different security system)

And it is possible that you might need to install xwayland on one or both machines.

dqrw, 2024-10-05:
I just performed 3 tests.
In the first test, I unplugged the dummy HDMI plug and powered up the RPi5. Then, from my main system, I used the "ssh -Y" command and logged into the RPi5. I changed into the proper subdirectory and executed "LinuxCNC SherlineMill.ini". The graphics of the LinuxCNC did display and I was able to run the default LinuxCNC logo program. But, the update rate of the graphics was pretty jerky when moving the logo around and the cutter pictured seemed to stall and then would jump ahead in its cutting path often. I guess this would be useable in most cases.
For the second test, I powdered up the RPi5 with the HDMI plug installed. I used TigerVNC on my main system to IP into the RPi5. I opened a terminal window on the RPi5, changed into the same subdirectory as before and executed the same "linuxcnc SherlineMill.ini" command. The same graphics for LinuxCNC logo appeared but the update rate was very smooth and responsive without any jerky appearance when moving the logo as well as when the cutter was in motion. This is how I was hoping to use the RPi5 without the dummy HDMI plug.
For the third test, I from my main system, I used the "ssh -X" command and logged into the RPi5. I then  executed the same test as test #1 and got the exact same results as test #1.
BTW, with the HDMI plug removed and attempting to log into the RPi5 with TigerVNC the error TigerVNC gave is:
"Failed to connect to (IP of the RPi5)unable to connect to socket: Connection refused (111)"
Whatever that means :-)
BTW (again) I had to give to trying to figure out how to apply the patches in the link you provided. I just am not well versed in how to use them. It seemed to imply I needed to rebuild the kernal somehow and that is definitely something I do not know how to do.
Please let me know if any additional information is required and I will do my best to provide it.  

Yahoo Mail: Search, Organize, Conquer 

The ssh -Y ***@***.*** (or whatever) will give you a remote terminal prompt (executing on the Pi).