LinuxCNC/linuxcnc #2978: Issue Parsing G4 Commands with Invalid Parameter

sfroome, 2024-05-15:
## Here are the steps I follow to reproduce the issue:

 1. Load a file that has a deliberate typo ( G4 X2)
 2. Press Start.
 
## This is what I expected to happen:
LinuxCNC should throw an interpreter error when the file is either loaded or attempts to run. 

Either the program is not allowed to run at all, or it throws a fault when it reaches the invalid line.

## This is what happened instead:

LinuxCNC does print a warning through gcode graphics upon loading the file and again upon running the program, but internally either an interpreter error is thrown and immediately reset or the interpreter error is never thrown. This means that the program proceeds when I press start, but then gets stuck on the line with the typo. 

While MDI commands and jogging can be performed after it gets "stuck", the bigger issue is the spindle remains turned on.

## Information about my hardware and software:

* Gcode used:

``` 

G1 X0 Y0 U0 V0 F100

G1 X10 Y10 U10 V10
M3
G4 X2

G1 X100 Y53
M2

```

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Ubuntu 22.04.4  
 * I am using this kernel version (shown by `uname -a`): 
 * I am running a binary I built myself
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): LinuxCNC 2.9.2
 * I am using this user interface (GUI): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa 7i96S

rmu75, 2024-05-16:
With lookahead for path optimization, the interpreter consumes more or less the whole file immediately and of course chokes on the G4 line and exits with an error. Up to this point motion segments are appended to a queue and emcmot is dutifully ececuting that queue until it is empty. I'm not sure if it is possible to retrieve the error in this case.

Perhaps the GUI should prevent running known-bad G-Code files, in other words, lock the run button if generation of preview yields INTERP_ERROR.

satiowadahc, 2024-05-17:
status.interpreter_errcode successfully shows the error after loading a file, however it seems the error gets cleared after a few seconds.

andypugh, 2024-05-17:
I wonder if the error flag clearing depends on the UI in use?

satiowadahc, 2024-05-17:
Axis, QtDragon, Our custom one, all have the same issue.

sfroome, 2024-05-21:
> With lookahead for path optimization, the interpreter consumes more or less the whole file immediately and of course chokes on the G4 line and exits with an error. Up to this point motion segments [...]

The GUI preventing running bad G-Code files would be the preferred outcome. LinuxCNC does appear to clear the error too fast in order to use it though. 

To add, I did add some debugging statements in LinuxCNC to trace the error. The result value in parse_file (in gcodemodule.cc) does get set equal to 5 (aka INTERP_ERROR), but it seems like the interpreter_error is never really handled by emctask.

revjmoney, 2026-11-09:
I traced this on master and I think the cause is in task, not the GUI. I'd like to work on a fix if you agree with the direction.

`G4 X2` has no P, so it fails `check_items` (`NCE_DWELL_TIME_MISSING_WITH_G4`, `interp_check.cc`), which runs from `parse_line` inside `Interp::read`. So this is a **read** error, not an execute error, and task treats the two differently in `emctaskmain.cc`:

- **Execute error** (`execRetval > INTERP_MIN_ERROR`): `interp_list.clear()` and `emcAbortCleanup(EMC_ABORT::INTERPRETER_ERROR, "interpreter error")`, which stops motion and the spindle.
- **Read error** (`readRetval > INTERP_MIN_ERROR`): handled together with `INTERP_ENDFILE` / `INTERP_EXIT` / `INTERP_EXECUTE_FINISH`. `print_interp_error` shows the message (the transient error @satiowadahc saw), then `interpState = WAITING` and nothing else. The lines already queued run out, and the program just stops at the bad line with M3 still in effect, as reported.

The `\todo MGS FIXME` right above that test already says it "actually evaluates to if (readRetval != INTERP_OK)".

Two ways to fix it, and I'd like your preference before I write it:

1. **Abort immediately on a read error**, the same as the execute-error path. Simple and consistent, but because of read-ahead it aborts before the good lines in front of the bad one have run.
2. **Abort when execution reaches the bad line**: remember the read error, let the queued motion drain, then call `emcAbortCleanup` (spindle off, coolant off, error stays visible). Closer to what the report expected ("throws a fault when it reaches the invalid line"), a bit more code.

Either way I'd add a regression test under `tests/` with this program. I have a 2.9 run-in-place build on Linux to test on. The GUI idea (disable Run when the preview reports INTERP_ERROR) still seems worth doing separately, but it would not cover programs sent without a preview.

grandixximo, 2026-11-10:
IMO option 2 is better: stop after the last good line has run, with the error still shown. Stopping well before the bad line is confusing when debugging, because nothing near the stop point is wrong.

Execute errors are also found at read-ahead time, so I think the deferred abort should cover both read and execute errors; otherwise the two paths still behave differently.

On blocking Run in the GUI for programs with errors, I lean permissive. A warning is fine, but I'd rather not get in the operator's way: they may know what they are doing. When I'm the operator I want the control, and when it's someone else... well, why did you run this program?

revjmoney, 2026-11-10:
Thanks @grandixximo. Option 2 it is: let the good lines run, then stop at the bad one with the error still shown. I'll make the deferred abort cover execute errors found at read-ahead too, so both paths behave the same, and add a regression test with this program. I'll leave the GUI alone (no blocking Run).

grandixximo, 2026-11-10:
@revjmoney for the test, could you extend tests/motion-logger/basic instead of adding a new directory? It already runs every .ngc in that directory through task in AUTO mode and diffs the motion commands against expected.<name>, so a g4-no-p.ngc with this program plus its expected log is enough to show the missing abort and spindle stop.

One limit: the mock motion there doesn't execute the moves, so it can't tell "abort now" from "abort after the queue drains". If you want the drain timing covered as well, tests/abort/g64 samples real motion with halsampler and could be a model, but I'd only go there if the motion-logger case isn't enough.