LinuxCNC/linuxcnc #2983: Broken homing error handling! If one joint fails to home the other joints will continue anyways.

JimmyRigIt, 2024-05-20:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:
My machines, z homes first. Then x any y together. Noticed this issue when my z stepper wire broke. 

 1. Unplug the stepper wire going to the  joint which homes first. In my case z axis
 2. Home the machine.
 3. Linuxcnc will throw an error saying it failed to home z but.....
 4. X and y axis will start homing after the error shows up causing the head to crash. 

## This is what I expected to happen:
Z joint tries to home, fails, pops up a warning saying z joint failed to home. Machine then stops. 

Their should be an option in ini file to disable this function for those without homing switches.

## This is what happened instead:
Machine crashes head cause the z axis was not moved up before x and y due to broken stepper motor wire

## It worked properly before this:
Don't think this worked
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

Seems to be an issue across linuxcnc. Tried it on a few machines and was able to replicate. Other users are reporting the same issue.
https://forum.linuxcnc.org/plasmac/48236-qtplasmac-2-10-arc-ok-thcad-thc-and-thcad-ohmic-troubleshooting-help-needed?start=40

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

JimmyRigIt, 2024-05-20:
Might have found the issue. 

in the homing.c file the "home_abort" only aborts the current joint not all of them. 

        case HOME_ABORT:
            H[joint_num].homing = 0;
            H[joint_num].homed = 0;
            H[joint_num].joint_in_sequence = 0;
            joint->free_tp.enable = 0;
            H[joint_num].home_state = HOME_IDLE;
            H[joint_num].index_enable = 0;
            immediate_state = 1;
            break;

petterreinholdtsen, 2024-05-21:
[JimmyRigIt]
> Might have found the issue. 
>
> in the homing.c file the "home_abort" only aborts the current joint
> not all of them.

Thank you for looking into this.

What do you believe would be the correct action, in this and the generic
case?

-- 
Happy hacking
Petter Reinholdtsen

JimmyRigIt, 2024-05-22:
the code i wrote above wasn't doing it, Phillc54 seems to have come up with a solution, i'll test on my machines over the next few days and report back. We are unsure of all the implications this could have (if any). Ideally this works with no side effects and that's all the corrective action needed.

[phillc54]
Something like this seems to work but I have not comprehensively tested for any side effects.
    /* check for reached end of move */
    if (! (&joints[jno])->free_tp.active) {
        /* reached end of move without hitting switch */
         (&joints[jno])->free_tp.enable = 0;
        rtapi_print_msg(RTAPI_MSG_ERR,_("j%d end of move in home state %d, homing aborted"),jno, H[jno].home_state);
        for (int j = 0; j < all_joints; j++) {
            if (!H[j].homed) {
                H[j].home_state = HOME_ABORT;
            }
        }
        return 1; // abort reqd
    }

JimmyRigIt, 2024-05-23:
That code is working on my machine! No ill effects so far

PR #3481 (2025-07-09): Abort homing on all joints when encoutering an error while homing any single joint
fixes #2983

There are 2 issues fixed with this PR:
1) The HOME_ABORT state in the homing state machine only aborts the homing of a single joint while other joints continue to travel, leading to potential crashes.
2) In all but one case (in the HOME_FINAL_MOVE_WAIT state), despite an error being reported, the state moved to  HOME_IDLE instead of HOME_ABORT.