LinuxCNC/linuxcnc #2987: Gcode with M67 commands give inverted result

royka1, 2024-05-28:
## Here are the steps I follow to reproduce the issue:

 1. Use LinuxCNC built from Master
 2. gcode made by Lightburn with M67 E0 Qxx commands in G90 mode
 3.

## This is what I expected to happen:
To give the output shown in Lightburn

## This is what happened instead:
When I laser engrave filled text in scanning mode it gives the inverted result. But when I replace all Q0 with Q30 and replace the Q30 with 0 it engraves correctly, except the overscanning part (2.5% of the beginning and end).

## It worked properly before this:
The same gcode works correctly when I install 2.9.2 again.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian Bookworm
 * I am using this kernel version (shown by `uname -a`): Linux orangepi5 5.10.160-rt89+ #8 SMP PREEMPT_RT
 * I am running ...
   * [ X ] A binary I built myself
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): Master built today.
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa 7I92m

andypugh, 2024-06-08:
So, are you saying that this is not an issue in the released 2.9.2 version, but you see this problem when you compile from source? 

My initial feeling is that this isn't a problem in the LinuxCNC code, but in your setup. I am not aware of any recent changes in the handling of the M67 code. 

What is the logical path between the M67 command and your laser? Is there a pwmgen involved? Which gpio driver are you using?

royka1, 2024-06-08:
> So, are you saying that this is not an issue in the released 2.9.2 version, but you see this problem when you compile from source?

Yes before I used a self compiled version 2.9 which worked correctly, after that I compiled 2.10 and the problem started. 
> 
> My initial feeling is that this isn't a problem in the LinuxCNC code, but in your setup. I am not aware of any recent changes in the handling of the M67 code.

Initially I was also thinking that it was a problem in my setup so tried about everything I could think of with the software and hardware, eventually I reverted back to the released version 2.9.2 and it's working correctly again.
> 
> What is the logical path between the M67 command and your laser? Is there a pwmgen involved? Which gpio driver are you using?

I'm using a Mesa 7I92m driver board with a 74HC04 as buffer/inverter
In the HAL file I have:
setp hm2_7i92.0.pwmgen.00.output-type 1
setp hm2_7i92.0.pwmgen.pwm_frequency 20000
setp hm2_7i92.0.pwmgen.00.scale 100
setp hm2_7i92.0.pwmgen.00.out0.invert_output True

net laser_pwm motion.analog-out-00 => hm2_7i92.0.pwmgen.00.value
net laser_pwm_enable motion.digital-out-00 => hm2_7i92.0.pwmgen.00.enable
net laser_pwm_enable => hm2_7i92.0.pwmgen.00.enable

andypugh, 2024-06-10:
setp hm2_7i92.0.pwmgen.00.out0.invert_output True

Might be relevant. 

But can you check if the first stage of the output is different in the two cases? 
This will be, I think, the motion.analog-out-00 value. I would be very surprised (and concerned) if that is different in the two versions.

royka1, 2024-06-13:
In this video you can see that V2.10 picks the m67 command from the next line.

https://github.com/LinuxCNC/linuxcnc/assets/397414/60cf42e9-6353-4572-b302-d7b0d00b8c56

royka1, 2024-06-17:
I've also tried the deb package from the linuxcnc site with the same result, on v2.10 it doesn't work correct.

Here a video of v2.9.2 with the same gcode which output the correct values:

https://github.com/LinuxCNC/linuxcnc/assets/397414/4577c279-198e-4461-a7c0-a332ae8127a9

royka1, 2024-06-17:
From this commit it doesn't work correctly:
https://github.com/LinuxCNC/linuxcnc/commit/392b0883718fd68434f2806c1b58d69759d94278

royka1, 2024-06-18:
After reverting commit https://github.com/LinuxCNC/linuxcnc/commit/6d6538695779eaf39c635caa51762fd4fa29b8a2 by downloading the patch: 
https://github.com/LinuxCNC/linuxcnc/commit/6d6538695779eaf39c635caa51762fd4fa29b8a2.patch
Somehow I had to do this command twice: git apply -R ../6d6538695779eaf39c635caa51762fd4fa29b8a2.patch 
It still compiled and LinuxCNC works correctly again.

rmu75, 2024-06-19:
This is strange -- I will look into it.

rmu75, 2024-06-20:
@royka1 can you try https://github.com/rmu75/linuxcnc/tree/rs/std-interp-list-fix this should fix it

royka1, 2024-06-20:
Yes that fixed it, thank you!

closed 2024-06-21 by royka1