LinuxCNC/linuxcnc #2995: Remove StateTag tag. As it is never used.

krcwrhvgyxzqelljes, 2024-06-11:
/home/user/linuxcnc/src/emc/rs274ngc/interp_convert.cc

Line 4357 

StateTag tag;

This can be deleted as it is never used.

andypugh, 2024-06-11:
In which version? 
In master it is: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_convert.cc#L4357
In 2.9 it is https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/rs274ngc/interp_convert.cc#L4357

rmu75, 2024-06-21:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/rs274ngc/interp_convert.cc#L3811
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_convert.cc#L4352

andypugh, 2024-06-22:
OK, yes, that seems to never be used in that scope. 
The compiler probably ignores it, but it does seem untidy.

rene-dev, 2024-07-07:
closed by https://github.com/LinuxCNC/linuxcnc/pull/2999

closed 2024-07-07 by rene-dev