LinuxCNC/linuxcnc #2996: URL not fining ISO file

derek084, 2024-06-14:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

andypugh, 2024-06-14:
Which URL? Where?

derek084, 2024-06-17:
Hi Andy,

using this script -

zsync https://www.linuxcnc.org/iso/linuxcnc_2.9.2-amd64.hybrid.iso

Returns

failed on url https://www.linuxcnc.org/iso/linuxcnc_2.9.2-amd64.hybrid.iso
could not read control file from URL
https://www.linuxcnc.org/iso/linuxcnc_2.9.2-amd64.hybrid.iso

Derek

NOTE:
1.  The contents of this email and any attachments are confidential and
privileged.  Any unauthorised use of the contents is expressly prohibited.
If you receive this email in error, please contact the sender and then
delete the email.
2.  Before opening or using any attachments, please check them for viruses
and defects.  The sender does not accept any liability for any damage
caused by such viruses and/or defects.  Please notify the sender of any
virus and/or defect should you discover same.
3.  This email is also subject to copyright.  No part of it should be
reproduced, adapted or communicated without the written consent of the
copyright owner.  Any personal information in this email must be handled in
accordance with the Privacy Act 1988 (Cth).

andypugh, 2024-06-19:
This looks like a capitalisation difference that was introduced when another developer updated the ISO image. 

zsync https://www.linuxcnc.org/iso/LinuxCNC_2.9.2-amd64.hybrid.iso 

Will probably work.

andypugh, 2024-06-19:
Sorry, slightly struggling to sort this out properly.

derek084, 2024-06-19:
All sorted  thanks

andypugh, 2024-06-27:
I found a secondary problem, it seems to only work with http rather than https. 
Odd, but now fixed I think.

closed 2024-06-27 by andypugh