LinuxCNC/linuxcnc #3000: Safety Issue - Spindle Starts itself after program pause, spindle stop, program stop 2.9

FriedrichKegel, 2024-07-02:
## Here are the steps I follow to reproduce the issue:
 1. Start a program with the spindle turning
 2. Pause the program
 3. Stop the spindle during the halt (e.g. for checking surface finish etc.)
 4. Stop the program

## This is what I expected to happen:
Spindle stays off

## This is what happened instead:
Spindle starts turning again?!

## It worked properly before this:
Upgrade from 2.8 to 2.9 

## Information about my hardware and software:
 * I am using this Linux distribution and version Debian 12
 * I am using this kernel version (shown by `uname -a`): 6.1.0-18.rt-amd64 #1 SMP PREEMPT_RT Debian6.1.76-1 (2024-02-01)
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version 2.9
 * I am using this user interface (GUI) Gmoccapy
 * I am using this interface hardware vendor and chipset Mesa 7i76e

rmu75, 2024-07-03:
#2764 may be related

andypugh, 2024-07-03:
I have tried to reproduce this behaviour, and failed. 
https://youtu.be/KTBlwkrh99E
This is running in a VM with the 2.9.2 ISO installed on it. 

Interestingly Axis does not allow you to stop the spindle when paused.

FriedrichKegel, 2024-07-04:
I checked your vid, but not quite the same - you continued the programm with a turned off spindle and stopped it then.
Here is the behavior in video: 
https://youtu.be/rDHB-4NIuYM

If you can let me know what to check from here, happy to do so!

Sigma1912, 2024-07-04:
There has been a similar report regarding unexpected restart of the spindle in Gmoccapy:
https://forum.linuxcnc.org/gmoccapy/51000-gmoccapy-spindle-restarts-at-program-stop#288177

rmu75, 2024-07-04:
FWIW I can't reproduce this issue. @FriedrichKegel I would check the HAL wiring and the tool measuring routine. If spindle.0.on is off and the spindle is running that is a misconfiguration.

andypugh, 2024-07-04:
Can you check spindle.0.speed-out ? 

My guess is that your system is configured such that spindle.0.on is ignored and spindle.0.speed (and friends) is enough to run the spindle. 

I won't go as far as to claim that this is a misconfiguration, I feel that speed-out should be zero too in this state. 

But, I still can't reproduce this behaviour. Can you try running the gmoccapy sim config and see what you see? 

Does your tool measurement routine use M70 / M72 ? Perhaps this is exactly #2764

andypugh, 2024-07-13:
@FriedrichKegel Can you confitm if your tool change remap uses M70/M72 ?

FriedrichKegel, 2024-07-16:
Hi together, I went through my config again and set up everything from scratch and got it working now. Sorry for the trouble here!
Unfortunately I can't report back from where exactly the error came -.-

closed 2024-07-16 by FriedrichKegel