LinuxCNC/linuxcnc #3025: Add missing manual pages for remaining binaries (melfagui millturngui qtplasmac_gcode setup_designer tool_mmap_read tool_watch xyzab-tdr-gui z_level_compensation modcompile)

petterreinholdtsen, 2024-07-11:
The lintian check in Debian report that some programs in /usr/bin/ are missing manual pages: melfagui millturngui qtplasmac_gcode setup_designer tool_mmap_read tool_watch xyzab-tdr-gui z_level_compensation modcompile.

Please add manual pages for these.

Other lintian related issues are#745, #924, #1728, #1917 and #1964.

andypugh, 2024-07-14:
Is it as simple as adding minimal pages with links to the main docs? If so, then please see the PR above.

petterreinholdtsen, 2024-07-14:
[Andypugh]
> Is it as simple as adding minimal pages with links to the main docs?
> If so, then please see the PR above.

It is a good start, at least.  Someone can add more content later, if we
want more in them.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2024-07-14:
Is that a recommendation to merge?

petterreinholdtsen, 2024-07-14:
[andypugh]
> Is that a recommendation to merge?

Sure.  I have not enough knowledge to judge the content, but see that
there seem to be sensible man page headers and that it will provide some
information to the man page readers.  Good enough for me. :)

-- 
Happy hacking
Petter Reinholdtsen

hansu, 2024-07-16:
Can this be closed or are some man pages remaining missing?

petterreinholdtsen, 2024-07-16:
[Hans Unzner]
> Can this be closed or are some man pages remain missing?

I belive everything was fixed in 2.9.  Have not yet seen the merge in
master, which I plan to convert to adoc before closing this issue.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2024-07-16:
I tried to do the merge to master last night, but it was difficult. (for reasons not to do with the docs. Issues with interp_g7x.cc and emcrsh mainly) 
Also I was trying to test compile in an RTAI system, and that has been broken too ( https://github.com/LinuxCNC/linuxcnc/commit/87e4e4428a1d8012dc021596b02f4b9085641232#r144266518 )