LinuxCNC/linuxcnc #3031: Ascidoc man pages are not shown on website

hansu, 2024-07-16:
The man pages which have been recently converted to adoc (https://github.com/LinuxCNC/linuxcnc/pull/1376) are not show on the website (http://linuxcnc.org/docs/devel/html/) for the master branch. 
There are still the previous man pages shown which are generated from the groff sources.
On a local build they are shown correctly.

Example to check:
https://linuxcnc.org/docs/devel/html/man/man1/io.1.html does have "SEE ALSO" but the sources does not have it: https://github.com/LinuxCNC/linuxcnc/blob/master/docs/src/man/man1/io.1.adoc

petterreinholdtsen, 2024-07-18:
[Hans Unzner]
> The man pages which have been recently converted to adoc
> (https://github.com/LinuxCNC/linuxcnc/pull/1376) are not show on the
> website (http://linuxcnc.org/docs/devel/html/) for the master branch. [...]

I have been told that the web pages are updated by a buildbot job, which
show the commands executed in the build log.  I tried to locate them,
but there seem to be problems with the buildbot installation.
<URL:http://buildbot.linuxcnc.org/buildbot/builders/0000.checkin/builds/10755/steps/garbage-collect%20local%20git%20repo/logs/stdio>
show 

  fatal: sha1 file './objects/pack/tmp_pack_7OvBfh' write error: No space left on device
  error: failed to run repack

Perhaps the problem is a full disk?

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2024-08-03:
According to @SebKuzminsky the problem was the infinite loop in po4a. 
I thought that @petterreinholdtsen had a workaround for that, but it might have been reverted.

petterreinholdtsen, 2024-08-04:
[Andy Pugh]
> According to @SebKuzminsky the problem was the infinite loop in po4a. 
> I thought that @petterreinholdtsen had a workaround for that, but it
> might have been reverted.

See d82caa4a1669e07c4b0b41f1c52ac54db4c7d1ad for relevant pointers.
@SebKuzminsky reverted the triggering change, while I dropped
translations for the problematic files.  The change we kept were the one
from @SebKuzminsky, while w wait for a proper po4a fix.

-- 
Happy hacking
Petter Reinholdtsen

closed 2024-09-27 by hansu