LinuxCNC/linuxcnc #3032: Many formatting errors after man poage conversion to adoc

hansu, 2024-07-17:
Since https://github.com/LinuxCNC/linuxcnc/pull/1376 (which makes the man pages being generated from adoc sources, wich is a great improvement!) 
some of them have not ben converted correctly.
**So please do not update the man pages until they are fixed.**

Here is a list of the differences of the man9 man pages:

[image]

It's mostly when bold formatting is involved:

[image]

[image]

Unfortunately due to https://github.com/LinuxCNC/linuxcnc/issues/3031 it is currently not possible to inspect the new generated man pages on the web site. You only can check this on a local build.

petterreinholdtsen, 2024-07-17:
[Hans Unzner]
> It's mostly when bold formatting is involved:

While some bold formatting were removed intentionally, the
classicladder.9 ones you show seem to be unintentintional.

I could have spend more time comparing the before and after versions,
but we wanted to complete the migration before the LinuxCNC developer
gathering were over and decided to fix any required cleanup
afterwords.  Very glad to see you are able to assist in tracking down
any problems. :)

-- 
Happy hacking
Petter Reinholdtsen

hansu, 2024-07-28:
The problem are strings composed of `<bold><italic>[<bold>]<regular>`
For example
**classicladder.0.in-**_NN_ IN bit::
should be 
`**classicladder.0.in-**_NN_ IN bit::`
but was converted like this
`*classicladder.0.in-*_NN_ IN bit::`

Also some asteriskes appeared before the double colon 

See here the fix for the first man pages: https://github.com/LinuxCNC/linuxcnc/pull/3045/files

@petterreinholdtsen  maybe we can fix this by applying an intelligent regular expression on it?

hansu, 2024-07-28:
See https://asciidoc-py.github.io/userguide.html#_unconstrained_quotes

petterreinholdtsen, 2024-07-28:
[Hans Unzner]
> @petterreinholdtsen  maybe we can fix this by applying an intelligent
> regular expression on it?

Feel free to try.  I used pandoc to do the conversion, and hoped it
would do a good enough job.  :)

hansu, 2024-07-28:
>Feel free to try.  

I was hoping you can help with your expertise ;-)

>I used pandoc to do the conversion, and hoped it
would do a good enough job.  :)

It seems that pandoc has some problems with the inline formatting. Maybe you can report that?

hansu, 2024-07-29:
@petterreinholdtsen  I have now fixed a few man pages and I think you could also fix some.

Otherwise I tend to revert the converted man pages because the goal was to simplify the maintenance of the man pages and not to introduce a huge amount of work to fix the errorneous converted man pages.

andypugh, 2024-08-03:
Is this fixed now? 
http://linuxcnc.org/docs/stable/html/man/man9/classicladder.9.html 
Looks OK to me.

hansu, 2024-08-03:
50% of the man9 pages are fine (double checked), the rest probably as well.

But you can't check it this way unfortunately because of https://github.com/LinuxCNC/linuxcnc/issues/3031

smoe, 2024-08-17:
I think we are mostly stuck with the man pages PRs of mine still failing on sid.

andypugh, 2024-09-30:
#3031 is showing as closed now?

hansu, 2024-09-30:
Yes, the docs are up to date on the website :tada:

andypugh, 2024-09-30:
So, what needs to be done to close this issue?

hansu, 2024-10-01:
Spent some hours in manually comparing man pages before and after # 1376.
I will have some several hours train journeys end of october, maybe I have then time to face this.

hansu, 2024-10-01:
I don't know how to give an AI two lists of files....
I could imagine that an intelligent or AI based diff tool would work.