LinuxCNC/linuxcnc #3037: COMPONENT PIN NAMES ARE GETTING TOO LONG

bennyb1, 2024-07-22:
## Problem with long  names 
 Some components have long pin or component names or both.
 
[image]

## This is what I expected to happen:
Programmers should refrain from using long names.
Developers should try to follow IEC 1131 standard for names.

## This is what happened instead:
I use an schematic capture software to draw hal based schematics, then run a custom script to convert the schematic into a hal file.
Long names makes the component function blocks large which consumes valubale cad sheets space.

## Information about my hardware and software:
I am using Autodesk Eagle for schematic caputure and a built in customized script to generate hal file.

jethornton, 2024-07-22:
What does IEC 1131 have to do with this project? IEC 1131 is for PLC languages.

The basic principle of IEC-1131 is that a programmer can develop a control algorithm (referred to as a "Project") using any combination of five control languages; Instruction List, Structured Text, Ladder Diagram, Function Block Diagram, and Sequential Function Chart.

rmu75, 2024-07-23:
YOU CAN IMPLEMENT THAT A MAPPING BETWEEN SHORT AND LONG PIN NAMES IN YOUR SCRIPT THAT GENERATES THE HAL FILE. E.G. USE "FIPS" IN YOUR SCHEMATIC AND HAVE THE SCRIPT TRANSLATE "FIPS" TO "FEED-INCHES-PER-MINUTE" IN YOUR HAL FILE.

closed 2024-07-26 by andypugh

bennyb1, 2024-07-27:
Thank you Robert for your suggestion to map pins to keep component names and pin names short. 

Regards,

Ben Benissa