LinuxCNC/linuxcnc #3057: Man pages in adoc format are not converted proberly

hansu, 2024-08-05:
Fir example line breaks and links are just ignored.

The conversion is not new, but we never had that many man pages in the adoc format so that it never has been noticed.

Maybe someone can point me to the command where this is done in the Submakefile? @petterreinholdtsen maybe?

petterreinholdtsen, 2024-08-06:
[Hans Unzner]
> Maybe someone can point me to the command where this is done in the
> Submakefile? @petterreinholdtsen maybe?

It is the a2x call at the end of docs/src/Submakefile.

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2024-12-06:
@hansu, where do we stand with the man page formatting? It is (mostly) fine now?

hansu, 2024-12-06:
I am currently waiting for a feature to be implemented in asciidoctor to be able to use that instead a2x for converting the man pages from adoc to groff.
That would save us to fix several man pages and also allow a proper formatting.

andypugh, 2025-02-03:
See also "[Emc-developers] RPi5 support on download homepage"
It seems that converting ascidoc tables to groff and then to HTML doesn't work. 
This can be seen by comparing
http://linuxcnc.org/docs/stable/html/man/man9/hm2_spix.9.html
to
https://github.com/LinuxCNC/linuxcnc/blob/master/docs/src/man/man9/hm2_spix.9.adoc

BsAtHome, 2025-02-04:
@hansu, which asciidoctor feature are you waiting for?

hansu, 2025-03-26:
> [@hansu](https://github.com/hansu), which asciidoctor feature are you waiting for?

This one: https://github.com/asciidoctor/asciidoctor/issues/4637

BsAtHome, 2025-03-26:
Couldn't that be "fixed" by using sed in post-processing?

hansu, 2025-03-26:
Have now given up waiting for asciidoctor and spend a lot of time to fix the formatting errors and make it as good as it is possible with our asciidoc-py.
Still not finished - but done with all man9 pages and the man1 pages from a-h.

hansu, 2025-03-26:
> Couldn't that be "fixed" by using sed in post-processing?

I tried to fix that via the Makefile, but didn't succeed. But I am not very experienced in makefiles. Feel free to try. It's basically just this change in the Makefile: https://github.com/hansu/linuxcnc/commit/f9cecb511142dd3b8c0527c09403f5b8437e1f7f
(and you need to install asciidoctor)

BsAtHome, 2025-03-26:
I'll have a look at it.

BsAtHome, 2025-04-05:
Only sideways. Been busy with real life and another project. But I haven't forgotten.

BsAtHome, 2025-06-02:
> Did you have a look at it?

Now I had a (real) look at it.

There are a few things that need to be take taken care of when moving to `asciidoctor`.
A few processing scripts need to find the aliases (already done) and do the fixup.

The Makefile dependency check differs between `a2x` and `asciidoctor`. Asciidoctor will use the header's section designator as extension for the generated man-page. For example, when the file reads:
``` adoc
= util_name_whatever(section)
...
```
then "section" would normally be `1`, `3` or `9` or, in case of HAL `3hal`, RTAPI `3rtapi` and hostmot2 uses `3hm2`. This needs to be expressed in the filename because asciidoctor generates filenames based on the specified section. A reasonable amount of files in docs/src/man/man3 have the wrong name (e.g. `hal_type_t.3.adoc` needs be renamed `hal_type_t.3hal.adoc`, see alternative below). The build will try to remake these pages if the files in question are not renamed.
 
Almost all manual pages from section 3 have a suffix that is more than the numeral `3`. Those that belong to `3` need to be examined where they actually belong. Or, see alternative below.
 
Installing man-pages should adhere to the system standard and that is different depending target distro. Also, the man-pages in section 3* are not necessarily searchable are because that requires an update of /etc/manpath.config (on Debian) or /etc/man_db.conf (on Fedora/Redhat) and add the relevant suffix in an additional SECTION line or you need to have a local ~/.manpath (or ~/.man_db) file to override for the local user. Neither sound appealing to me.

Much better alternative is to drop the non-numerical part in section 3 and use purely a single digit `3` as section. That would require some more edits in the files, but would make it simpler and wouldn't require OS level man config changes. This change would be more in line with the old `a2x` way, where most (alias) pages would belong to section 3, regardless what the source said. A2x apparently looks at the `:manvolnum:` and uses that as file suffix.
I'm not sure why we would need `3hal`, `3rtapi` and `3hm2` anyway because all functions described and documented are prefixed with `hal_`, `rtapi_` and `hm2_` anyway. Therefore, there will be no collisions. In that case, I'd suggest to remove "undocumented.{3hal,3rtapi}.adoc" because these are, well, not very informative. There also seems to be "skeleton.{3hal,3rtapi,9}" as leftovers from old times that should be removed.

The other use of `a2x` should be much easier to handle, I think. It is used to make PDF documentation. However, that I have not tested with `asciidoctor`.

petterreinholdtsen, 2026-01-06:
[BsAtHome]
> Much better alternative is to drop the non-numerical part in section 3
> and use purely a single digit `3` as section.

I agree that this is the better alternative, and also recommend renaming
all 3* sections to 3.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2026-01-06:
I have a feeling that the 3hal (etc) is what sorts the manpages into the sections at the bottom here:
https://linuxcnc.org/docs/stable/html/