LinuxCNC/linuxcnc #3061: Issue with pncconf script with "string" to "int" convert 7i95T

ustal, 2024-08-11:
Hi, a problem with the unexpected output from the reading card (during Pncconf wizard):

[image]
[output.txt](https://github.com/user-attachments/files/16571159/output.txt)
[mesa0_discovered.xml.txt](https://github.com/user-attachments/files/16571160/mesa0_discovered.xml.txt)

The same with <secondaryinstance>MUXEDQCOUNT</secondaryinstance> instead of int

So for the first issue should be smth like "First letter" instead of "any position" temp[0] == 'P', or better regexp like P\d
For the second I have no idea what should be there.

c-morley, 2024-08-18:
Is this from pncconf's discovery option?
This problem seems unsolvable with reasonable work.
'IO Connections for Step/DIR+Serial+Encoders -> 7I95_0' is a new way to describe connectors.
Traditionally Mesa used P plus a number.

I have lost interest in chasing differences in firmware descriptions.
The XML files which were supposed to be the machine readable version was much the same.

It's probably time to finally remove pncconf from linuxcnc.

ustal, 2024-08-18:
TBH pncconf make better settings for antoher board, which I could refactor than MesaCT. MesaSupport asked me to use pncconf for 7i96 to setup spindle, because MesaCT for 7i95(T) doesn't setup spindle properly (step/dir and PWM). Also Pncconf use pid for  servo and step motors instead of stepgen (examples in manual for LinuxCNC). So I would like to ask you do not remove this tool because it's useful.

Yes, that was errors during discovery. One of issues I fixed just replace "if" statement for looking "P" letter in IO Connector. Another related to <secondaryinstance>MUXEDQCOUNT</secondaryinstance> instead of integer inside.

c-morley, 2024-08-18:
Pncconf has been around along time so has lots of 'learned lessons' in it which makes it useful when it works.
It's been adapted/stretched a long way from the original code base and it shows here and there.
it hurts me to have a semi broken utility with my name on it :)
The part that makes the INI/HAL works pretty well, it's the part that reads the firmware/displays the choices that needs a rethink.

I won't be removing it anytime soon then, but don't expect to do anything but keep it run-able and simple fixes.

can you show me the exact code you used to fix the one error?

ustal, 2024-08-21:
1194 instead of looking letter P in temp, you could use either regex or just looking first char and do not include regex lib
 if 'P' in temp[0]: