LinuxCNC/linuxcnc #3064: Tool table wiodget: expand for column "rotation"

hansu, 2024-08-11:
[image]

Sigma1912, 2024-08-12:
I agree. This makes much more sense as each WCS has it's own 'R' value.

andypugh, 2024-08-13:
And having a spurious rotation is an occasional cause of trouble. (I saw a recent forum post where someone said that their X and Y axes had swapped. That was the cause.

c-morley, 2024-08-18:
Does each system have it's own rotation? I thought the rotation was was separate and added after the user system.

Sigma1912, 2024-08-19:
Yes, each system has it's own rotation. AxisGUI displays R values correctly for each system.

hansu, 2024-08-19:
Yes, except G92. See https://linuxcnc.org/docs/devel/html/gcode/g-code.html#gcode:g10-l2

c-morley, 2024-08-19:
What python attribute gives the rotation setting for systems not in use?
rotation_xy only gives the current rotation.

Sigma1912, 2024-08-19:
As far as I'm aware there are only python attributes for for the currently active WCS system.  I think one would have to query parameters #5230 + i*20, i=[0..8]

c-morley, 2024-08-19:
Ahh yes ,it's coming back to me now. We read the var file directly for the other user systems. I assume there is a var in that file for the other rotations too.

hansu, 2024-08-26:
@c-morley Are you currently working on this? If yes - I thought it would be nice if we could use the calculator widget to enter the offset values. What do you think?

hansu, 2025-06-15:
Fixed by https://github.com/LinuxCNC/linuxcnc/pull/3457