LinuxCNC/linuxcnc #3068: master branch. Luck of probe_material_z function / versa_probe

chernogorsky, 2024-08-15:
## Here are the steps I follow to reproduce the issue:

 1. Enable versa_probe
 2. Go to probe tab
 3. Select check block Z height

## This is what I expected to happen:
It starts working

## This is what happened instead:
It has an error "function probe_material_z not in probe routine"

## It worked properly before this:
it works in 2.9.3

and it has that function in 
probe_routines.py

When def probe_material_z(self): copied from 2.9.3 and inserted into /usr/lib/python3/dist-packages/qtvcp/widgets/probe_routines.py
it works as expected

## Information about my hardware and software:
 * I am running ...
   * [X] A binary I built myself (from git master branch)

c-morley, 2024-08-16:
Looks like it was accidental with last push - I'll fix it soon. Sorry

c-morley, 2024-08-16:
Should be fixed now.

closed 2024-08-16 by c-morley