LinuxCNC/linuxcnc #3074: QtDesigner and `qtvcp` app crashes when using `TabWidget` from `Linuxcnc - HAL` section

Ornstein89, 2024-08-17:
# QtDesigner and `qtvcp` app crashes when using `TabWidget` from `Linuxcnc - HAL` section

## Here are the steps I follow to reproduce the issue:

 1. Install LinuxCNC with `*.iso` from https://www.linuxcnc.org/iso/, tried both `linuxcnc_2.9.3-amd64.hybrid.iso` and `linuxcnc_2.9.1-amd64.hybrid.iso`.
 2. Install qt development tools:
    ```bash
    sudo apt-get update
    sudo apt-get install qttools5-dev-tools qttools5-dev libpython3-dev
    ```
 3. Install qt designer plugin:
   `bash /usr/lib/python3/dist-packages/qtvcp/designer/install_script` with admin password and option `3 = Yes, for a package installation`.
 4. Launch Qt Designer:
   `designer -qt=5`
 5. Try to drop `TabWidget` from `Linuxcnc - HAL` section of `Widget Box` to the form, look screen record video https://www.youtube.com/watch?v=aZdqGYQfb6U, LinuxCNC 2.9.3 to the left, LinuxCNC 2.9.1 to the right:
    
    <div align="left">
      [image]
    </div>

 7. Qt Designer immediately crashes with following terminal print out (same for Linux CNC 2.9.3):
    ```bash
    pc1@linuxcnc291:~$ designer -qt=5
    Qtvcp python plugin found: /home/pc1/.designer/plugins/python/qtvcp_plugin.py
    [DEFAULT][INFO]  Logging to: /home/pc1/default.log (logger.py:106)
    tool_mmap_user(): tool mmap not available
    poll(): continuing without tool mmap data
    link (updating variable file): No such file or directory
    link (updating variable file): No such file or directory
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/qtvcp/widgets/tab_widget.py", line 43, in tabSizeHint
        return QSize(size.width(), w)
               ^^^^^^^^^^^^^^^^^^^^^^
    TypeError: arguments did not match any overloaded call:
      QSize(): too many arguments
      QSize(w: int, h: int): argument 2 has unexpected type 'float'
      QSize(a0: QSize): argument 1 has unexpected type 'int'
    Aborted
    ```
    
  8. Also one of my existing qtvcp applications, which worked fine on LinuxCNC 2.8.x and contains `qtvcp`→`Linuxcnc - HAL`→`TabWidget`, crashes with same print out:
   
    ```bash
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/qtvcp/widgets/tab_widget.py", line 43, in tabSizeHint
        return QSize(size.width(), w)
               ^^^^^^^^^^^^^^^^^^^^^^
    TypeError: arguments did not match any overloaded call:
      QSize(): too many arguments
      QSize(w: int, h: int): argument 2 has unexpected type 'float'
      QSize(a0: QSize): argument 1 has unexpected type 'int'
    Aborted
    ```

## This is what I expected to happen:

I expect I could design and use `qtvcp` applications with `Linuxcnc - HAL` `TabWidget` component.

## This is what happened instead:

Qt designer and `qtvcp` app crash immediately when using `Linuxcnc - HAL` `TabWidget` component.

## It worked properly before this:

Worked properly in LinuxCNC 2.8.x (2.8.1-2.8.4)

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
   * Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version (shown by `uname -a`):
   * `Linux linuxcnc293 6.1.0-23-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux`
   * `Linux linuxcnc291 6.1.0-13-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux`
 * I am running
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
   * `2.9.1`, `2.9.3`
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): `qtvcp`.
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): parallel port.

Ornstein89, 2024-08-17:
Oh, I see appropriate changes in https://github.com/LinuxCNC/linuxcnc/blob/05047eebcc38cb0d8960838ac7bc2d815eab13ee/lib/python/qtvcp/widgets/tab_widget.py#L43 already. I'll try to switch to buildbot version and close if problem solved.

Sigma1912, 2024-08-17:
Looks like these changes should also have gone into 2.9 and not only master:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/lib/python/qtvcp/widgets/tab_widget.py

c-morley, 2024-08-18:
Thank you for the report. I have pushed the fix to 2.9.

closed 2024-08-18 by c-morley