LinuxCNC/linuxcnc #3079: emccanon function signature changed causing error message

c-morley, 2024-08-18:
recent master

running sim/qtdragon/qtdragon_probe_tool/tdragon_auto_tool_probe.ini
causes an error message when loading chips.ngc:

 G-Code error in 3D_Chips.ngc
Near line 0 of
/home/chris/linuxcnc/nc_files/examples/3D_Chips.ngc
M6/change_epilog: Python argument types in
    emccanon.CHANGE_TOOL(int)
did not match C++ signature:
    CHANGE_TOOL(void)

This config uses remap of m6

c-morley, 2024-08-18:
ok this is caused because I have an installed version of linuxcnc that is 2.9 based.
My sim INI uses paths like: ~/linuxcnc/nc_files/examples/remap_lib
I do this so that in an installed version the sim actually works.

Any idea how to best handle the need for different paths for RIP vrs installed versions?

6XoCtujg2C0gne, 2024-08-19:
Hi Chris,

In ~/linuxcnc-dev/scripts/rip-environment

 

At the very top is a CASE statement as follows:

 

  27 case "$0" in

  28     rip-environment|*/rip-environment)

  29         as_command=true

  30         if [ $# -eq 0 ]; then

  31                 cat <<-EOF

  32             This script can be used in one of two ways.

  33

  34             It can be loaded in the context of your shell, by executing

  35                 . $0

  36             after this, commands like 'linuxcnc', 'halrun' and so on refer to the

  37             version in this directory, instead of to an installed version of

  38             linuxcnc (if any)

  39

  40             Second, it can be used to run a command in this directory without

  41             modifying the environment of the calling shell:

  42                 $0 command...

  43             such as

  44                 $0 halrun foo.hal

  45                 $0 linuxcnc configs/.../foo.ini

  46         EOF

  47     exit 1

  48         fi ;;

  49     *) as_command=false

  50 esac

 

I’d suggest creating a variable to support the RIP environment or something along those lines.

KennethThompson, 2025-03-05:
> ok this is caused because I have an installed version of linuxcnc that is 2.9 based. My sim INI uses paths like: ~/linuxcnc/nc_files/examples/remap_lib I do this so that in an installed version the [...]

Chris. I am the author of a plugin for Rapid Change ATC that uses qtdragon_hd. M6 is remapped properly, and the first tool pickup works great. But then when a second tool is selected either manually or through a program's gcode, an error appears with this exact reference to the signature of emccanon.CHANGE_TOOL(int) did not match C++ signature CHANG_TOOL(void). Based on your comments in this thread, it appears to be an issue with pathing of the python glue file.  But the answer from 6X about the rip-environment does not appear applicable.  Do you have any suggestions on how to fix this signature issue?  @c-morley

c-morley, 2025-03-07:
Can you post the remap code? And you can confirm only linuxcnc 2.9 is on the system (not master too) ?
Are you saying you want a custom version of Qtdragon to supply to your customers? yes it's possible.
If you can supply the custom panel I'll see what I can do.
Nice looking tool changer and good job on the custom panel!

KennethThompson, 2025-03-07:
> Can you post the remap code? And you can confirm only linuxcnc 2.9 is on the system (not master too) ? Are you saying you want a custom version of Qtdragon to supply to your customers? yes it's [...]

I will see about getting answers to your questions. I've been working with a user of my plugin who realized he was running the latest, 2.10, of linux cnc and then downgraded.  We are working through a different error now.   I have this plugin released under the MIT license, and would love to just bundle it with qtdragon if its possible. The plugin is specific to the RapidChange ATC.  I'll get my sim profile in a git repo with the plugin so you can have a copy that loads up without error.

closed 2026-05-10 by BsAtHome